stefanbode/Sonoff-Tasmota #168: Device does not accept cmnd via MQTT set command via Client ==> err msg always "Shutter unknown"

markus-fehling, 2019-07-28:
Team, I need some guidance .... I have quite a few tasmota devices, but I do not get this on to accept cmnd send via MQTT clients.
FullTopic: /SmartHome/Buero/%topic%/%prefix%/
Topic: ROLLO
Client: BueroRollo_%06X
cmnd via console works: like
/SmartHome/Buero/ROLLO/cmnd/POWER1 off
/SmartHome/Buero/ROLLO/cmnd/POWER1 off
/SmartHome/Buero/ROLLO/cmnd/shutterposition 20
but if I use the MQTT client I get back always "Shutter unknown" ...
Thanks for help, Markus

stefanbode, 2019-07-28:
Hi Markus, this should work. All commands you send as command will also work on MQTT. I assume that you did not use the correct topic. This is the most likely reason. Please send a status 0 report and we will have a look

markus-fehling, 2019-07-29:
Hi Stefan, thanks a lot for the promp respone.
It tuned out that the issue is with the MQTT.fx client .... I chaned it to another one iand it worked ... 
-- Markus

closed 2019-08-18 by stefanbode