stefanbode/Sonoff-Tasmota #169: Sonoff Basic compiling error

titomontes, 2019-08-01:
Hello Stefan

I´m trying to compile your fork to flash it onto a Sonoff dual R2, but I´m getting only errors; It seemed (like 20 attempts ago) that the problem would be low memory, so I tried the sonoff basic option, but I get this:
` Compiling .pio\build\sonoff-basic\src\sonoff.ino.cpp.o
Linking .pio\build\sonoff-basic\firmware.elf
c:/users/tito/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
 .pio\build\sonoff-basic\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\sonoff-basic\firmware.elf] Error 1
 [ERROR] Took 17.14 seconds
Environment sonoff-basic        [FAILED]`

Is there an already compiled flashable .bin file? (with core 2.3?)

Thanks again for your work!

ascillato, 2019-08-01:
@titomontes 

Your issue is that you have Platformio v4 and requires to add `build_dir = .pioenvs` under **[platformio]** inside the _platformio.ini_ file as:

```ini
[platformio]
src_dir = sonoff
build_dir = .pioenvs
```

This is already updated in main Tasmota

meingraham, 2019-08-01:
@stefanbode @titomontes 

Have you tried using Gitpod for compilation? It ought to work in a very similar manner to what is described in the [Tasmota wiki Gitpod article](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) other than pointing to your code branch.

Mike

stefanbode, 2019-08-01:
Hi gents, thanks for bringing up the incompatibility with Platformio V4. Will update sources now

closed 2019-08-01 by stefanbode

titomontes, 2019-08-09:
Hello again

I´m sorry to say that, compiling in core 2.3 still fails: 

> Linking .pioenvs\sonoff\firmware.elf
> .pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
> .pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
> .pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
> core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
> core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
> collect2.exe: error: ld returned 1 exit status
> 
> *** [.pioenvs\sonoff\firmware.elf] Error 1
>  [ERROR] Took 32.14 seconds
> Environment sonoff              [FAILED]

Core 2.5.2 works well though, although there are stability issues...

titomontes, 2019-08-09:
> @stefanbode @titomontes
> 
> Have you tried using Gitpod for compilation? It ought to work in a very similar manner to what is described in the [Tasmota wiki Gitpod article](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) other than pointing to your code branch.
> 
> Mike

@meingraham Thanks a lot for pointing it out. I will surely use it in the future

titomontes, 2019-08-13:
@ascillato @stefanbode, did you have a look at this?

stefanbode, 2019-08-13:
Yes fine. I think you can use it without any modification. I do not see any  changes I need to do. Anyhow: I do not agree to change the my_user_config.... you can all do this in the override file. Just follow the exampled and undef define the variable to override existing values.