stefanbode/Sonoff-Tasmota #172: mismatching UI

discip, 2019-08-03:
Good afternoon everyone,
after having flashed a shelly2.5 with the firmware provided by Stefan Bode (https://github.com/stefanbode/Sonoff-Tasmota/blob/master/bin/sonoff.bin) and configuring it according to his wiki, I am encountering a confusing UI, where the positions of the buttons do not correlate to its slider counterpart (Please refer to the attached screenshot.).
[image]
No matter how I am changing the template, even using the "SHUTTERINVERT 1"command, the UI stays the same.
Meaning: When sliding to the left, the right button is activated and vice versa.
So, it would be nice, if either the buttons or the slider are getting flipped horizontally.
Or am I missing something?
Thank you in advance.
Kind regards

stefanbode, 2019-08-04:
Works as designed. Maybe shutterinvert will fix this. No change planned

closed 2019-08-04 by stefanbode

discip, 2019-08-05:
Good afternoon Mr Bode,
first of all: Tank you very much for this nice piece of work of yours!

Sorry for being persistent, but I think, I did not make myself clear enough.
The thing is about the UI exclusively.
So far I did not encounter any other issues with your firmware. In fact I am very happy and thankful, that you did this. 
As depicted in the attached screenshot, the label 'Close' is above the button "Toggle 1" and the label 'Open' is above the button "Toggle 2".
No matter how I change the settings in the template the UI will not change at all. Even using the "SHUTTERINVERT 1"command, did not solve this.
'Toggle 1' is always activated when the slider is being moved to the right and
'Toggle 2' is always activated when the slider is being moved to the left.
My point is:
The action executed by the slider at its left most position, should correspond to the action executed by the button on the left side and
the action executed by the slider at its right most position, should match the action executed by the button on the right side.
I hope this time I made my self clear enough. ;)
Thank you in advance.
kind regards