stefanbode/Sonoff-Tasmota #173: MQTT commands for blinds

vidalpascual, 2019-08-05:
I'm trying to integrate a Sonoff Dual R2 with your Tasmota fork in HomeKit, using this Homebridge plugin: https://www.npmjs.com/package/homebridge-mqtt-blinds

The problem is that I don't know the available MQTT commands in this Tasmota fork for the blinds. I need to know what I need to put in these three lines:

Variable | Description | Example
-- | -- | --
currentPosition | Topic to get the current position of your blind. The value has to be between 0-100. | GET/currentPosition
positionState | Topic to get the current position state of your blind. Custom PositionStates messages from your device can be configured with mqttPositionStateValues. | GET/positionState
targetPosition | Topic to get the target position of your blind. The value has to be between 0-100. | GET/targetPosition

Can you help me? Thanks!

stefanbode, 2019-08-05:
target shutterposition will be propagated IN RESULT e.g.:
hm/status/ESP_001E92/RESULT       {"shutterposition1":100}
or after shutterstop in
hm/status/ESP_001E92/SHUTTER1 100

positionState is not supported

vidalpascual, 2019-08-05:
It would be possible to implement positionState to know if blind is closing, opening or stopped?

I don't know how Home Assistant implement blinds without this parameter... :/

vidalpascual, 2019-08-05:
Maybe it's possible to use the parameter 'direction' for this, right? 1 for opening, -1 for closing and 0 for stopped. Is there a MQTT command to get this value?

21:03:14 MQT: tele/sonoff_dual_1/RESULT = {"SHUTTER-1":{"position":49, "direction":0}}
21:03:15 Wait for Motorstop 500
21:03:16 MQT: stat/sonoff_dual_1/RESULT = {"POWER1":"ON"}
21:03:16 MQT: stat/sonoff_dual_1/POWER1 = ON
21:03:16 Shutter 0: Real Pos: 10100, Target 20000, source: Shutter, start-pos: 49 %, direction: 1, rtcshutter: 0.2  [s]
21:03:17 MQT: stat/sonoff_dual_1/RESULT = {"POWER1":"OFF"}
21:03:17 MQT: stat/sonoff_dual_1/POWER1 = OFF
21:03:17 MQT: stat/sonoff_dual_1/SHUTTER1 = 60
21:03:17 MQT: tele/sonoff_dual_1/RESULT = {"SHUTTER-1":{"position":60, "direction":0}}
21:03:20 Wait for Motorstop 500
21:03:20 MQT: stat/sonoff_dual_1/RESULT = {"POWER2":"ON"}
21:03:20 MQT: stat/sonoff_dual_1/POWER2 = ON
21:03:26 Shutter 0: Real Pos: 400, Target 0, source: Shutter, start-pos: 60 %, direction: -1, rtcshutter: 5.8  [s]
21:03:26 MQT: stat/sonoff_dual_1/SHUTTER1 = 0
21:03:26 MQT: stat/sonoff_dual_1/RESULT = {"POWER2":"OFF"}
21:03:27 MQT: stat/sonoff_dual_1/POWER2 = OFF
21:03:27 MQT: tele/sonoff_dual_1/RESULT = {"SHUTTER-1":{"position":0, "direction":0}}

FauthD, 2019-08-28:
Hi,
Perhaps this helps.
Between 6.1 and 6.5 there have been changes to the behavior.

1. The tele, cmnd etc used to be in front of the rest, now it comes after: %prefix%/%topic%/ now %topic%/%prefix%/ (I have seen that on original Tasmota as well.)
2. Topics SENSOR and RESULT now send the Direction additionally to the position. This requires a change to the parsing template. See my V6.5 example below.

Note: The SENSOR comes with quite some delay.

I first write my setup for V6.5 used on my cover on south side.
Then I add the older setup for V6.1.1 on the west side cover.
These lines go into cover.yaml in Home Assistant.
V6.5:
```
- platform: mqtt
  name: "Büro Süd"
  device_class: "window"
  command_topic: "shutter/buero/sued/cmnd/Backlog"
  availability_topic: "shutter/buero/sued/tele/LWT"
  qos: 0
  retain: true
  payload_open: "SHUTTEROPEN"
  payload_close: "SHUTTERCLOSE"
  payload_stop: "SHUTTERSTOP"
  state_open: "ON"
  state_closed: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  position_topic: "shutter/buero/sued/tele/SENSOR"
  value_template: '{{ value_json["SHUTTER-1"]["position"] }}'
  position_open: 100
  position_closed: 0
  set_position_topic: "shutter/buero/sued/cmnd/shutterposition"
```
V6.1.1:

```
- platform: mqtt
  name: "Büro West"
  device_class: "window"
  command_topic: "cmnd/shutter/buero/west/Backlog"
  availability_topic: "tele/shutter/buero/west/LWT"
  qos: 0
  retain: true
  payload_open: "SHUTTEROPEN"
  payload_close: "SHUTTERCLOSE"
  payload_stop: "SHUTTERSTOP"
  state_open: "ON"
  state_closed: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  position_topic: "tele/shutter/buero/west/SENSOR"
  value_template: '{{ value_json["SHUTTER-1"] }}'
  position_open: 100
  position_closed: 0
  set_position_topic: "cmnd/shutter/buero/west/shutterposition"
```

MadDoct, 2019-08-29:
@FauthD , sorry but that simply isn't true... The topic keeps on beeing %prefix%/%topic%/ , unless you change it...

serbasan, 2019-08-29:
I am also interested in being published by mqtt if the blind is opening, closing or stopping. Thank you.

stefanbode, 2019-08-30:
Hi all, made some changes yesterday to optimize battery usage. I found some interesting option to support the rules in tasmota. I will add some some events that you can use in rules to send data via mqtt

stefanbode, 2019-08-30:
Shutter now support RULES. You can now send your own messages. Please see the wiki for more help. This should solve your problem. Let me know if more is required. I can also help on rules if required.

closed 2019-08-30 by stefanbode

FauthD, 2019-08-31:
Hi MadDoct,
The reason for the changed order in %prefix%/%topic%/ is probably that I enabled Home Assistant discovery. I have seen some comment just yesterday that said this changes the order.

onkeldittmayer, 2019-09-25:
is it possible to send Shutter MQTT Events by Script?
I have MoesHouse 2 Gang Relay (https://www.aliexpress.com/item/33047521274.html) which use frequency recognition for the switches (see: https://github.com/arendst/Sonoff-Tasmota/issues/6085 )
However, to solve this problem it is necessary to use Script instead of Rules in Tasmota.
I compiled your fork (Thanks for your great work!!) with Script enabled and it work fine, but Shutter don't send the shutterposition1 value when using the hardware-switch.

stefanbode, 2019-09-25:
Hi, I never worked with scripts so far. I assume there is also an adition required like in rules to make if fully functional. Normally tasmota is smart enough to transfer all these messages to the different positions in the code. Can you share some script code that should work but do not? This might make it easier to find the root cause.

stefanbode, 2019-09-25:
can you check if the newly implemented variable sht[ for shutter helps. It should report the shutter position similar to the pwr[ command to get the power state. This was the easy part. I still have no idea how to set the shutter position :-)

stefanbode, 2019-09-25:
ah ok maybe this work to set a shutterposition:
=> shutterposition1 35

Let me know if you need more,

onkeldittmayer, 2019-09-25:
thanks for your fast answer and suport.
I'm a noob and have to read AND understand a lot.
I did not get a working script which send the current Shutter position via MQTT. :-(

But i noticed the shutterposition will also NOT send via MQTT when i use the WEB Buttons.
Can you confirm this issue on an untouched master image?

stefanbode, 2019-09-26:
You should get the position at the end of the movement in a MQtt command similar to this one
 MQT: hm/status/ESP_001F3E/SHUTTER1 = 100 (retained)

This works whatever you do to move. Also webbuttons  do the work.

onkeldittmayer, 2019-09-26:
Yep, semms the Sonoff Module in ioBroker can't handle the ShutterPosition.
I tryed another MQTT Module and viola the Variable change the Value in Broker on every use. Now it would be perfect to get the direction on an movement (seems to be transmitted on TelePeriod) and i have to figure out how to send the value from the other Module to the Tasmota Client.

onkeldittmayer, 2019-10-01:
So i finaly find the problem:
i got an MQT massage when i use the slider:
18:28:34 MQT: stat/%topic%/RESULT = {"shutterposition1":100}
when i use the Buttons or switches this Massage don't appear.

when I manualy command: Publish stat/%topic%/RESULT {"shutterposition1":0}
the Sonoff Module on IOBroker axapted that.
is it possible to implement this in your code?
would be great for all noobs (like me) to use your shutter code ;-)

and a direction MQT like stat/%topic%/RESULT {"direction":0}
where 0 can be 1 or -1 for up / down / stop would pe perfect.