stefanbode/Sonoff-Tasmota #174: Interworking of Deepsleep and Teleperiod

thomas-lentz, 2019-08-16:
Hi Stefan, first a short but big "Thank You" for your excellent work!

### ISSUE DESCRIPTION - TROUBLESHOOTING
Before asking I checked your Wiki, but couldn't find any detailled Information. Here what I could get and what I couldn't get:

- **deepsleep=xxx [sec, 15s<=xxx<=136 Years]**
xxx defines the time in seconds the device goes to sleep until it wakes up again (without external triggers).

- **teleperiod=xxx [sec, 10s<=xxx<=300s]**
Intervall the sensor data is reported 

### REQUESTED INFORMATION

1. How is the interworking between the timers? 
2. The device will wake up after _deepsleep seconds_, will it then check if the time elapsed after last sending the sensor data is longer ago than _teleperiod seconds_?
3. The sleep time (device in deepsleep) and the wake time are both counted towards the teleperiod intervall, right?
4. Are these two examples correct?
 - deepsleep=60, teleperiod=120
The device wakes up and sends every second wakeup the sensor data is sent (a potential shift of the two intervalls neglected)
 - deepsleep=60, teleperiod<=60
The device wakes up and sends sensordata everytime it wakes up
5. What is the condition for the device for going back to sleep?
3. The seconds for the current sleep intervall are started the moment the device goes to deep sleep, not with the wake up of the device after the end of the previous deepsleep intervall, right?

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

stefanbode, 2019-08-17:
Teleperiod get resetted with every deepsleep. It is not stored in rtc memory. If you set teleperiod to 10 the device will do a measurement ASAP and go back to sleep, if you set it to 20 then the device will wait 20 sec after restart to do the measurements. Letting other stuff stabilize. Then do the measurements and go to deepsleep. All my garden sensors work with teleperiod 10.

thomas-lentz, 2019-08-17:
Hi Stefan, thanks for the explanation and responding so fast.

thomas-lentz, 2019-08-18:
If the deepsleep period is smaller than the teleperiod period the device consequently never goes to sleep or am I wrong?

stefanbode, 2019-08-18:
wrong. deepsleep is always the sleep time and does not recalculate the time is awake. therefore is is nearly impossible to let the device wake-up every hour exact at the same time. An interesting feature, but not implemented now. the time between two measurements is deepsleep + online time. The last can vary a bit from time to time depending on how fast the network came up an e.g. the time-server responds.

thomas-lentz, 2019-08-18:
Okay, I understood - thank you.

closed 2019-08-18 by thomas-lentz