stefanbode/Sonoff-Tasmota #177: Stop roller blind by sending same mqtt shutterposition again.

cinemarene, 2019-08-20:
Is it possible that if you send the same position again, the roller blind will stop?

**Have you look for this feature in other issues and in the wiki?**
yes

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Send mqtt command eg. "shutterposition 60", if i send it again, then the the roller blind will stop?

stefanbode, 2019-08-26:
hi, this makes no sense in two reasons. First the firmware dows not log the last command, second why should a setposition command stop the shutter if it still is not there where it should be. You can send a shutterstop at anytime to stop the shutter. this is a clear command. if the shutter is stopped and you send shutterstop it will start moving..... no, just kidding.

cinemarene, 2019-08-26:
Okay. Thanks anyway.

closed 2019-08-26 by cinemarene