stefanbode/Sonoff-Tasmota #178: Calibration problem

andreamastro, 2019-08-24:
I have a problem with calibration. After some time 0% and 100% no longer match, so i can't open or close my shutter blind completely and i have to reset values each time. How can i fix it?

stefanbode, 2019-08-26:
hi, can you give some more information? There are already some "special functions" included that add extra runtime to ensure it will find the endstops. Please provide some more info.

andreamastro, 2019-08-26:
I have a sonoff dual R2 and this is what i did to configure It:
- SETOPTION80 1
- INTERLOCK 1
- SETOPTION81 1
- SHUTTEROPENDURATION, SHUTTERCLOSEDURATION, SHUTTER50PERCENT for calibration
- switchmode1 2, switchmode2 2 to handle switch pressing when shutter blind has been moved from webUI or Home Assistant

stefanbode, 2019-08-26:
With the current implementation of TASMOTA it might be that it is INTERLOCK 1,2 to put relay 1 and 2 into the interlock. What is exactly your problem? Do you loose calibation parameter? are your not able to close or not able to open?

andreamastro, 2019-08-26:
After some time that i use the shutter blind i can't completely open it, but when i close it completely i can again open it completely and this because percent no longer match with real position of the shutter blind

stefanbode, 2019-08-26:
for a first fix please add 1 second to shutteropenduration and tell us if it gets better/worse or don't change. I assume it will get better. regarding the INTERLOCK 1,2: You must be on a new version to use this. If you're still on 5.x this is also fine. no need to change.

andreamastro, 2019-08-26:
In this way when i open the shutter blind completely 50 percent position doesn't match with real 50 percent due to 1 second added

stefanbode, 2019-08-28:
Yes this is also common known problem that can be calibrated with the option SHUTTER50PERCENT.
Drive the shutter once to 50 percent (visually) and write down the value you see. e.g. 63. Then set SHUTTER50PERCENT 63.
Now 50% is 50% and the rest is more or less also fine. just between 0% and 10% there are by design bigger errors. Please take a look in the calibration section of the wiki: https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays

MadDoct, 2019-08-30:
If you need to correct the problem after some time, you can do the following:
Open the shutter to 100%.after it reaches that point, write in the console - backlog POWER1 on; delay 20; power1 off
And now it should be reset...

stefanbode, 2019-08-30:
This will normally not fix the problem, because the shutter software will switch it off immediately. As mentioned before OPEN and CLOSEDURATION must be long enough to open and close and the relation between these two must match to prevent shifting. Then SHUTTER50PERCENT to get it user-freindly.

MadDoct, 2019-08-30:
Sorry to disagree, but you are wrong @stefanbode... I would rather have it as you say, but truth is, if you issue a "power1 on" on the console, the relay will never turn off unless you turn it off... Maybe it is a bug, but it's how it works...

andreamastro, 2019-08-31:
> If you need to correct the problem after some time, you can do the following:
> Open the shutter to 100%.After it reaches that point, write in the console - backlog POWER1 on; delay 20; power1 off
> And now it should be reset...

But this is not a definitive solution. I have to do this every time that the problem appears

udo1toni, 2019-08-31:
Every now and then, I drive the shutters to both end positions. 
I have one shutter, that shall not close entirely through the night, and this one tended to close more and more, so I increased uptime by one second (I think I should flash a current version with finer graduation), now the roller shutter won't close entirely when never opening complete, but as I always open the shutters to upper end, that's no problem...

stefanbode, 2019-08-31:
@MadDoct can you confirm that have executed both commands: INTERLOCK 1 and INTERLOCK 1,2
Tasmoto Supports interlock since some versions. The behavior you see is only possible if interlock is not recognized. Anyhow I will double check. Maybe a bug

MadDoct, 2019-08-31:
Yes @stefanbode, I do have interlocking enabled, and before that I had setoption14... ;)
So, if it isn't by design, it is a bug...

MadDoct, 2019-08-31:
(OFF-TOPIC) By the WAY, could you please review my pull request respecting to the pcf?

stefanbode, 2019-09-01:
No PCF will be no default because of large code. can be easily enabled wirh config. Can you send me you init line that the shutter is reporting. Should look like:

```
00:00:00 Shutter 0 (Relay:1): Init. Pos: 20000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc
: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
```

Im intrested in the shuttermode your device is reporting.

MadDoct, 2019-09-01:
You didn't read what I wrote correctly... If I use the web interface (the buttons that qre there) to activate the relay, it works as expected... It doesn't when you issue a "power1 on" on the console or using mqtt... Try it and report back, please...

stefanbode, 2019-09-02:
you're right. If you send the command through MQTT or Serial it has the "source" = I (ignore). This is quite bad. Other changes also come with "I" and here we should not interact. I need to make some investigations if and how this can be fixed.

MadDoct, 2019-09-03:
I'm not sure that "fixing" this would be such a good idea... The fact that it is as it is, allows you to easily fix calibration problems like the ones reported in the OP...

stefanbode, 2019-09-03:
I will do a YouTube video later this week, showing how to setup shutters. Maybe this convince me to add additional features.

stefanbode, 2019-09-03:
To overcome calibration issues you can drive shutter to 80% then send shuttersetclose and shutteropen this will 100% drive the shutter to the upper entstop.

bwired-nl, 2019-09-03:
Where can i find the video regarding the shutters once its done!
thanks!

electrontube, 2019-09-05:
This all seems to be related to 
https://github.com/stefanbode/Sonoff-Tasmota/issues/74

An issue that is as old as the SB Fork and that has never been really fixed.

@stefanbode 
we are looking forward to a finally working release...

stefanbode, 2019-09-07:
https://youtu.be/Z-grrvnu2bU 

I just uploaded a calibration video. It also shows, why 0.1 sec is a better calibration as 1sec. Enjoy and hopefully, it helps.

stefanbode, 2019-09-07:
@electrontube : There is no issue in the firmware that needs to be fixed. If so, I will do. I this special case just some people did not follow the instructions. Just simple. Hopefully the video will explain it better.

zwinsch, 2019-09-19:
...as mentioned in the other thread: the physical up/down speed relation is variable with position. So, as long as the 50% position is calibrated only for comming from 100% there will always be a "drift" if the shutter is moved several times up and down by 10%. It would make a difference, if the 50% position could be calibrated independingly for comming from 100% and from 0% - still not perfect, sure, but much better...

stefanbode, 2019-09-19:
This is not true in my mind. With the 50% calibration I just change the external reported position. Internally it is still 63% for example. Therefore you will reach the 50% no matter if you come from close or open. Regarding a slip of position please review my YouTube video I posted also in the wiki. This should explain straight forward how to make it very smooth. I have a blind over my balcony that follows the sun. Many small movements to open and one to close finally. All works like a charm.

stefanbode, 2019-10-01:
There is now a new callibrationmatrix that maps the % from SHUTTERPOSITION exactly to the %opening you can see. Described in the wiki. Positions are exact no matter if you come from open or close. Openduration and closeduration should not more than 1 second more than required.

closed 2019-10-09 by stefanbode