stefanbode/Sonoff-Tasmota #179: A Place for "Pre-DeepSleep" function calls

thomas-lentz, 2019-08-27:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I need to know the best place to put Pre-DeepSleep function calls.
Concrete example: I want to switch off a sensor and a display just prior Tasmota goes to sleep - otherwise these devices consume senselessly power during the deepsleep phase.

### REQUESTED INFORMATION
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Device used (i.e. Sonoff Basic) : LOLIN D1 mini Pro V2.0.0
- [x] Selfcompiled firmware version: Project sonoff Sonoff Version 6.6.0.3 stb-1.6(sonoff)-2_5_0
- [x] Development IDE - Compiler / Upload tools used : Arduino / Arduino
- [x] Provide the output of command ``status 0`` :

stefanbode, 2019-08-27:
There is currently no functionality for this

thomas-lentz, 2019-08-27:
Hi Stefan,
I just need to know, which function is called to enter the deepsleep state, the rest I can program myself (of course I would publish it here).
I would be something before or shortly after the

`18:10:11 MQT: tele/skybeam/UPTIME_S = {"Time":"2019-08-27T18:10:11", "Uptime_s":0}`
message.

stefanbode, 2019-08-28:
I made a small change and you can now place your function in FUNC_SAVE_BEFORE_RESTART. For Example you create you own driver e.g xdrv_50_mydriver.ino and in the bottom under.

```
bool Xdrv50(uint8_t function)
{
  bool result = false;

    switch (function) {
      case FUNC_SAVE_BEFORE_RESTART:
        mypoweroff();
        break;
     ....
```

Just take a look at the other drivers how they do it. The function will be called for drivers and sensors.

stefanbode, 2019-08-28:
Cool addition with the change is, that you now can also use rules. "System#Save" is the rule event that is triggered just before deepsleep.

thomas-lentz, 2019-08-28:
Great, thank you - that's what I needed.

closed 2019-08-28 by thomas-lentz