stefanbode/Sonoff-Tasmota #180: DisplayMode command result 128 (should be 0 to 5)

thomas-lentz, 2019-09-02:
### BUG DESCRIPTION
When issuing the **_DisplayMode_** command it results in 128 (should be 0 to 5)

### REQUESTED INFORMATION
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Device used (i.e. Sonoff Basic) : LOLIN D1 mini Pro V2.0.0
- [x] Selfcompiled firmware version: Project sonoff Sonoff Version 6.6.0.3 stb-1.6(sonoff)-2_5_0
- [x] Development IDE - Compiler / Upload tools used : Arduino / Arduino
- [x] Provide the output of command ``status 0`` :
```
17:45:07 CMD: status 0
17:45:07 RSL: stat/.../STATUS   = {"Status":{"Module":73,"FriendlyName":["...."],"Topic":"....l","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:45:07 RSL: stat/.../STATUS1  = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"...","OtaUrl":"...","RestartReason":"Software/System restart","Uptime":"0T00:03:38","StartupUTC":"2019-09-02T16:41:29","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":6,"SaveAddress":"3F6000"}}
17:45:07 RSL: stat/.../STATUS2  = {"StatusFWR":{"Version":"6.6.0.3 stb-1.9(sonoff)","BuildDateTime":"2019-09-02T18:38:40","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
17:45:07 RSL: stat/.../STATUS3  = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"...","LogPort":514,"SSId":["...","..."],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
17:45:07 RSL: stat/.../STATUS4  = {"StatusMEM":{"ProgramSize":621,"Free":3452,"Heap":23760,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":2,"Features":["00000809","0FEAE396","001683A0","22B6179D","01001BC0"]}}
17:45:07 RSL: stat/.../STATUS5  = {"StatusNET":{"Hostname":"...","IPAddress":"...","Gateway":"...","Subnetmask":"255.255.255.0","DNSServer":"...","Mac":"84:F3:EB:E3:C9:89","Webserver":2,"WifiConfig":4}}
17:45:07 RSL: stat/.../STATUS6  = {"StatusMQT":{"MqttHost":"...","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E3C989","MqttUser":"...","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:45:07 RSL: stat/.../STATUS7  = {"StatusTIM":{"UTC":"Mon Sep 02 16:45:07 2019","Local":"Mon Sep 02 17:45:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:31"}}
17:45:07 RSL: stat/.../STATUS10 = {"StatusSNS":{"Time":"2019-09-02T17:45:07","Epoch":1567442707,"DS18B20":{"Temperature":26.8},"BME280":{"Temperature":27.3,"Humidity":22.9,"Pressure":1009.7},"PressureUnit":"hPa","TempUnit":"C"}}
17:45:07 RSL: stat/.../STATUS11 = {"StatusSTS":{"Time":"2019-09-02T17:45:07","Epoch":1567442707,"Uptime":"0T00:03:38","UptimeSec":218,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":61,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"..","BSSId":"...","Channel":1,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:12","DeepSleep":0,"Heap":23720}}}
```
- [x] Provide the output of console when you experience your issue if apply :
```
CONSOLE OUTPUT HERE:
17:46:17 CMD: displaymode
17:46:17 MQT: stat/rcsg-pool/RESULT = {"DisplayMode":128}
```

### TO REPRODUCE

Just enter **_DisplayMode_** at console

### EXPECTED BEHAVIOUR
After initial flashing or entering **_reset 1_** I expect 
MQT: stat/rcsg-pool/RESULT = {"DisplayMode":1}

When changing the DisplayMode (0 to 5) I expect the DisplayMode I've chosen.

### ADDITIONAL CONTEXT

Activate:
  **#define USE_DISPLAY                            // Add I2C Display Support (+2k code)**
in
 **my_user_config.h**

before compiling

thomas-lentz, 2019-09-02:
Let me add this - this also worked before without any problems...

stefanbode, 2019-09-02:
Ok, first I did not make any changes in this module. Anyhow, I tried it and had to change first the Board to Generic and then define SCL and SDA pins. After this, the command displaymode was available and shows 0. I made a reset 1 and defined configuration again. All fine at my side.

thomas-lentz, 2019-09-02:
Sorry, I should have given more detail...
My display is working fine, of course SDA, SCL and OLED-Reset configured.
The only thing which is not working is switching display modes. First thing it doesn't show the default display mode (1), when it is initialized and then you can't switch the displaymode. This is why I think, that something happened to the TASMOTA settings management in this fork. Maybe in the enums. I can check it later myself, but first I have to finish this project. Maybe it is also dependend on the display driver (mine is SD1306)

stefanbode, 2019-09-02:
I have one Sd1306 here. Maybe I can give it a try. The eNums of the Display a separate handled in the drv file. So very unlikely. I will attach a display later this evening. Anyhow as I said. I already saw displaymode 0 after reset.

stefanbode, 2019-09-06:
It looks like this was a bug of the TASMOTA Version I used for the build. I now upgraded to .10 version. Now on my side displaymode is 1. Anyhow it does not work. Maybe 5V was to much.

closed 2019-09-09 by stefanbode