stefanbode/Sonoff-Tasmota #182: Teepao Roller/shutter switch with UP, DOWN and PAUSE buttons

deltabert, 2019-09-04:
Hi Stefan,
I try to use a Teepao Roller/Shutter Switch foor my window screens. I flashed your fork of the Tasmora fw, following the leads and cfg example from this forum:
https://community.home-assistant.io/t/teepao-window-cover-shutters-integration/114548

The switch is working fine for up and down! But the device has a middle button which functions with factory fw af a stop or pause function. 
Is there a way this could be implemented with your fork of Tasmota?
I tried also the origional Tasmota fw from Theo Arends, and this shown the 3d button in the web interface. (but i could not get this fw working properly, so reverted back to your fork.)
Hopefully you can help me with this...
Kind regards, Bert

stefanbode, 2019-09-04:
Hi Bert, do you need the Button "RELAY3" or do you want to make your physical switch "PAUSE" working. Because in this case, you need a BUTTON1.  if you connect the GPIO to GND then the event is executed.
Just enter one rule:
```
rule1 ON button1#state DO shutterstop ENDON
rule1 on
```
as soon as you disconnect BUTTON from ground the shutter stop

deltabert, 2019-09-05:
Hi Stefan, Thanks for your reaction. I am not sure what you mean with your answer.
So i did not tried the commands you provided yet.

The Teepao switch has 3 switches, an UP, an DOWN and in between the 3rd button with a Pause simbol. My intension is to use that 3rd button to be able to stop the movement of a screen after it has been started Up or Down.

There is some strange behaviour with this switch. Pressing the middle/Pause button stops the relais, but the way it does this is different if you first pressed the Down button than if you first pressed the Up button. (Hope i make myself clear...)
Example.  Starting position: Screen is up. Button Up (POWER1) process is finished), then I have these status msg from mqtt:

stat/Screen_1/POWER1 ON
stat/Screen_1/SHUTTER1 100
stat/Screen_1/RESULT {"POWER1":"OFF"}
stat/Screen_1/POWER1 OFF

When i now press the Down button and after a few seconds the Pause button once it lights red and i hear a relay clicking, but the Pause button remains red. I have to press the Pause button a second time in order to really stop the movement and then the Pause button turns blue. This is what i see in the mqtt status:

mosquitto_sub -h localhost  -u openhab -P ***** -t stat/Screen_1/# -v 
stat/Screen_1/RESULT {"POWER2":"ON"}
stat/Screen_1/POWER2 ON
   _**[Here i pressed the first time]**_
stat/Screen_1/RESULT {"POWER2":"OFF"}
stat/Screen_1/POWER2 OFF
stat/Screen_1/RESULT {"POWER1":"ON"}
stat/Screen_1/POWER1 ON
   _**[Now i pressed the second time]**_
stat/Screen_1/RESULT {"POWER1":"OFF"}
stat/Screen_1/POWER1 OFF
stat/Screen_1/SHUTTER1 74

What i see is by the first press a relay goes off and immediately on again.

This is different when i start with a closed screen (Button down (POWER2) completely finished).
When i press the button Up (POWER1) and after a few second i press the Pause button, then the switch stops. This is the status in mqtt:

mosquitto_sub -h localhost  -u openhab -P ***** -t stat/Screen_1/# -v 
stat/Screen_1/RESULT {"POWER1":"ON"}
stat/Screen_1/POWER1 ON
_**[Here i pressed the first and only time]**_
stat/Screen_1/RESULT {"POWER1":"OFF"}
stat/Screen_1/POWER1 OFF
stat/Screen_1/SHUTTER1 33

Actually, If this switch had only 2 buttons, it was fine already. But now with this 3rd button i have to do something with it in order to comply to the WAFactor...

Can you perhaps help me with this?

stefanbode, 2019-09-05:
Hi can you show me the GPIO mapping. As far as I understand you have RELAY1 and RELAY2. And then I assume BUTTON1 and BUTTON2 to switch on the relay up/down for the WAF factor. For the pause you need a BUTTON3 and implement a rule what should happen on pressing pause:
```
rule1 ON button3#state DO shutterstop ENDON
rule1 on
```
There was a smaller bug that the relay when triggered by serial or mqtt did not switch off automatically. Please ensure you have all these options set:
```
setoption80 1
interlock on
interlock 1,2
```

stefanbode, 2019-09-05:
I have added in the Shutter Wiki some code from the logfile how it should look like when the shutter is operating correctly

deltabert, 2019-09-06:
This is the GPIO mapping:
[image]

As what the  configuration concerns, it took this from Linus Dietz, at this website:
https://community.home-assistant.io/t/teepao-window-cover-shutters-integration/114548
Insert into the template fileds the following JSON data:
_Quote:_
{"NAME":"Taopao","GPIO":[255,57,255,18,22,19,255,255,58,21,59,255,17],"FLAG":1,"BASE":18}
Check the “activate” box and save settings, your device will now reboot.
Go to “Console” and type in the following settings:
SETOPTION80 1
SETOPTION81 1
INTERLOCK ON
INTERLOCK 1,2      - Important !
SLEEP 0      - Important !
_Unquote:_

I use this switch but reading your posts, i am not sure if this is the right config for the switch.
If i put your rule commands in the console i get these messages:
13:33:52 CMD: rule1 ON button3#state DO shutterstop ENDON
13:33:52 Shutter unknown
13:33:52 MQT: stat/Screen_1/RESULT = {"Command":"Unknown"}
13:34:23 CMD: rule1 on
13:34:23 Shutter unknown
13:34:23 MQT: stat/Screen_1/RESULT = {"Command":"Unknown"}

Not sure why it is complaining about Shutter unknown. Is it because i renamed my device Screen-1?
Cant remember what the name was before i changed is 
I'am sorry but i'am not  that familiar with the ins-and-outs of the Tasmota firmware...

deltabert, 2019-09-06:
Loaded a second switch with the Tasmota Fork and gave these commands in the console:
00:00:00 Project Shutter Screen-2 Version 6.5.0.3 stb-1.3(sonoff)-2_4_0
00:00:00 WIF: Connecting to AP1 NoAccess-3 in mode 11N as Shutter-1233...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on Shutter-1233 with IP address 192.168.178.182
14:07:04 CMD: SETOPTION80 1
14:07:04 RSL: stat/Shutter/RESULT = {"SetOption80":"ON"}
14:07:19 CMD: SETOPTION81 1
14:07:19 RSL: stat/Shutter/RESULT = {"SetOption81":"ON"}
14:07:36 CMD: INTERLOCK ON
14:07:36 RSL: stat/Shutter/RESULT = {"Interlock":"ON","Groups":"1,2"}
14:07:55 CMD: INTERLOCK 1,2
14:07:55 RSL: stat/Shutter/RESULT = {"Interlock":"ON","Groups":"1,2"}
14:08:10 CMD: SLEEP 0
14:08:10 RSL: stat/Shutter/RESULT = {"Sleep":"0 (0)"}
14:09:00 CMD: rule1 ON button3#state DO shutterstop ENDON
14:09:00 Shutter unknown
14:09:00 RSL: stat/Shutter/RESULT = {"Command":"Unknown"}
14:09:38 CMD: setoption80 1
14:09:38 RSL: stat/Shutter/RESULT = {"SetOption80":"ON"}
14:09:57 CMD: interlock on
14:09:57 RSL: stat/Shutter/RESULT = {"Interlock":"ON","Groups":"1,2"}
14:10:14 CMD: interlock 1,2
14:10:14 RSL: stat/Shutter/RESULT = {"Interlock":"ON","Groups":"1,2"}

Before i gave the console commands, i changed the IP@ AND the Friendly Name to Screen-2.
But also here the RULE commands give the same errors!

stefanbode, 2019-09-06:
two things that might be important. please first do a restart. Then the shutter gets initilized. with the command shutterrelay 1 you should see all parameters. please post here. the rule support is not working in your version. sorry for this. this is newly added.

deltabert, 2019-09-06:
OkStefan, this is the output from shutterrelay 1:
19:57:28 Shutterrelay 1: is:1
19:57:28 Shutter accuracy digits: 1
19:57:28 Shutter 0 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 96 , Max Way: 50000, Opentime 25.0 [s], Closetime 26.0 [s], CoedffCalc: c0: 0, c1 500, c2: 500, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
19:57:28 MQT: stat/Screen_1/RESULT = {"shutterrelay1":1}

I am using a bin file which i got from Linus Dietz. This is version:
Sonoff-Tasmota 6.5.0.3 stb-1.3 by Theo Arends/Stefan Bode

Would it be possible for you to send me a latest bin-file eg via wetransfer or some other cloud storage?

stefanbode, 2019-09-08:
I have uploaded new binaries to the /bin folder. Let me know if this works for you

deltabert, 2019-09-08:
Hi Stefan, this morning, a new attempt.
I already downloaded your fork into VSCode/PlatfornIO, compiled it and flashed to the Teepao switch.
Configured it as follows:
Template: {"NAME":"Teepao","GPIO":[255,57,255,18,22,19,255,255,58,21,59,255,17],"FLAG":1,"BASE":18}
Activated and rebooted. Then in Console:
SETOPTION80 1
SETOPTION81 1
INTERLOCK ON
INTERLOCK 1,2 
SLEEP 0 
Restarted the ESP and again in Console:
SHUTTEROPENDURATION 25
SHUTTERCLOSEDURATION 25
Restarted the ESP then in Console:
rule1 ON button3#state DO shutterstop ENDON
rule1 on
Restarted the ESP and tested it. The Pause button works now as it should! ;-)

With this version the shutterrelay command works:
12:57:51 CMD: shutterrelay 1
12:57:51 Shutterrelay 1: is:1
12:57:51 Shutter accuracy digits: 1
12:57:51 Shutter 0 (Relay:1): Init. Pos: 19500 [39 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 50000, Opentime 25.0 [s], Closetime 25.0 [s], CoedffCalc: c0: 0, c1 500, c2: 500, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0
12:57:51 MQT: stat/Screen_2/RESULT = {"shutterrelay1":1}

At the moment i am still testing the switches, because at the first attempt both switches where all of a sudden reverted back to the default settings; lost all cfg, mqtt, template etc...
Flashed them again, and now they are running for the past three hours OK...
Perhaps tomorrow i'l test your bin file, but there is still a great difference in file size between your new bin file and the bin file i got after compiling... And i left out also several options which i don't use with these switches... My filesize is 470KB while yours is 85KB...

stefanbode, 2019-09-08:
Where do you see the 85kb. All the files in the BIN directory have 560kb. roundabout. Regarding the loss of settings. I know this is pain in the ass and for me, as a fork of the master Tasmota it is most times impossible for me to keep compatibility. Arend has some "healing" when upgrading his versions. I'm still not able to do this also for my devices. As long as the TASMOTA version is on the same sublevel. e.g. 6.6 now the upgrade should work without a loss. But I'm pretty sure in 6.7 Arnd again will drive me crazy. It is quite tough to keep it in sync. The good thing now is that the code for the shutter is very stable for some month. It works for me and many others, too. Therefore a merge into master can now be an option. But I have to check with Arend, because this is not only a new INO file. I also need changes in the master code that needs to be discussed

stefanbode, 2019-09-08:
If your shutter device is up and running you should make a backup. There is a nasty function in TASMOTA that if a reboot is "not successful" the firmware will disable after each boot more settings and finally after reboot 6 do a complete reset of the device to defaults. There is value in this behavior because it is more unlikely you lost a device completely. With the decode.py you can convert the backup into something readable.

stefanbode, 2019-09-08:
Have you seen my shutter calibration video? I see your open and close duration are equal. https://www.youtube.com/watch?v=Z-grrvnu2bU

deltabert, 2019-09-09:
> Where do you see the 85kb. All the files in the BIN directory have 560kb. roundabout. 
I am sorry, but i went to https://github.com/stefanbode/Sonoff-Tasmota/tree/master/bin downloaded sonoff.bin and it is:
[image]

As what syncing your fork with Theo Arends project, i understand the difficulties. But when your release is stable enough i have no problem with keeping an 'special' version for the screens.

As what the backup concerns, yes after this error and installing everything again i immediately did made backups of the two switches ;-) It scared me however, because the relays in the switch were flipping constantly. This happened with both switches. 
I had already one switch installed in place and removed it, i thought i had broken the switch (short circuited or something while installing). 
But back on the workbench i found it had only lost its configuration.

My open and close duration settings are almost the ones they should have with one of my screens.
I watched you video and it explains very well what i have to do once the switches are in place.
Thank you so much for all your help so far!

But please come back on the file size of your bin file, perhaps i am doing something wrong with the download? I have not yet tried your new sonoff.bin file. I am using now you complete source which i downloaded yesterday, and so far it is still working. Not yet an reset of the config.

stefanbode, 2019-09-09:
There is definitely something going wrong with your download. I clicked on the link and it is: 567 KB and if I download it is still that size.

deltabert, 2019-09-09:
I found what was going wrong with the download.
I downloaded by going to the page with the sonoff.bin. Did a right click and choose save as.
Then it is saved as 85KB. If i just click on sonoff.bin, a new page opens and there is a button to download. Then it is the correct size.
Strange this difference in behaviour!   Sorry, i am not that familiar with GitHub...

closed 2019-10-09 by stefanbode