stefanbode/Sonoff-Tasmota #183: Compilation fail with last version (cca8a25)

calcipher49, 2019-09-08:
Hi,

I will to use last version of stefanbode fork, but compilation fail at last step using Visual Studio Code with Platform IO, enabling only default_envs = sonoff in plaformio.ini.

Error message as follows:
Linking .pioenvs\sonoff\firmware.elf
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff\firmware.elf] Error 1

Thanks if you can help

stefanbode, 2019-09-08:
Do you still have a zzz_debug.ino in your sonoff directory? this needs be deleted

stefanbode, 2019-09-08:
I have uploaded new builds from the current version, if this helps

calcipher49, 2019-09-08:
Hi Stefan

Compilation working nice with new current version, thanks very much for assistance.

I close this issue...

closed 2019-09-08 by calcipher49