stefanbode/Sonoff-Tasmota #184: NodeMCU: Interblocco e Configurazioni Miste

Eligio-Greco, 2019-09-12:
Ciao StefanBode,

scusa se scrivo in italiano, ho bisogno di aiuto.

Ho uno scenario con un balcone dotato "electric shutter", "electric awning" e "external light"

Mi piacerebbe poter comandare lo "shutter" e lo "awning" con i classici pulsanti su/giù e con Home Assistant tramite Tasmota.

Vorrei utilizzare un NodeMCU, una scheda relè a 8 canali, un piccolo alimentatore e tutti gli accessori necessari.

Ho provato a fare dei test sul NodeMCU ma ho qualche problema.

1) Mi piacerebbe configurare tasmota con: {"Interlock":"OFF","Groups":"1,2"}, mi sembra la configurazione elettrica migliore.

Tuttavia se configuro 2 "Button" per i pulsanti su/giù, un pulsante decide la direzione, l'altro fa da start/stop: 

È POSSIBILE CONSERVARE LA FUNZIONE DI INTERBLOCCO MA AVERE UN COMPORTAMENTO DEI PULSANTI NORMALE?

Impostando {"Interlock":"ON","Groups":"1,2"} funziona tutto correttamente ma si perde la sicurezza elettrica: se un relè si "incolla" sono guai !

L'unica soluzione che ho trovato è quella di usare {"Interlock":"ON","Groups":"1,2"} e aggiungere altri 2 relè collegati in questo modo:

[image]

Tuttavia mi sembra insensato avere un micro a disposizione e dover utilizzare 4 relè per comandare 1 motore.

2) Se configuro {"Interlock":"OFF","Groups":"1,2"}, quando si attiva il secondo relè per cambiare direzione resta attivato anche se il primo relè si disattiva: non è un problema ma ciò rappresenta uno spreco di energia inutile, quel relè potrebbe restare attivo anche per giorni senza azionare il motore, consumando inutilmente energia:

È POSSIBILE ATTIVARE IL SECONDO RELÈ SOLO QUANDO SERVE ?

3) Vorrei configurare il NodeMCU con:
- 2 relè e 2 "button" per l' "electric shutter" {"Interlock":"ON","Groups":"1,2"};
- 2 relè e 2 "button" per l' "electric awning" {"Interlock":"ON","Groups":"3,4"};
- 1 relè e 1 switch per la "light".

SI PUÒ FARE ?

Non appena configuro i button Tasmota si resetta alla configurazione originale (Sonoff Basic): perchè ?

Per Esempio:
--------------
D0 = Relè 1i - Shutter On/Off
D1 = Relè 2i - Shutter Up/Down
---------------------------------------
D2 = Relè 3i - Awning On/Off
D3 = Relè 4i - Awning Open/Close
---------------------------------------
D4 = Relè 5 - On/Off - Light
---------------------------------------
---------------------------------------
D5 = Button 1i - Shutter Up
D6 = Button 2i - Shutter Down
---------------------------------------
D7 = Button 3i - Awning Open
D8 = Button 4i - Awning Close
---------------------------------------
Rx = Switch 1 - Toggle Light

Grazie in anticipo per l'aiuto che vorrai darmi.

stefanbode, 2019-09-12:
English would make it much easier. The interlock 1,2 and other are not required and maybe a problem for the fail-safe configuration. I test on interlock and pulse timer to determine the operation mode. The image above does not look like fail-safe configuration. For me this looks like a normal shutter. Please see the wiki how to wire this configuration. There are images explaining

stefanbode, 2019-09-12:
You have to define the interlock in two steps:
INTERLOCK ON
INTERLOCK 1,2 3,4

Otherwise you overwrite and will not be happy

stefanbode, 2019-09-12:
Now google the translation... first delete all the interlock. Second wire the shutter as described in the failsafe configuration. Relay 1 will go off after movement by definition. On relay 2 you can define a pulsetimer with e.g. 180 seconds. This would do the job. Or you can add a rule on the event shutter moved and switch off relay 2

stefanbode, 2019-09-12:
Regarding the reset I also see sometimes strange behavior. For me it looks like the original firmware is a bit to nervous about wrong configurations. If there is more than one reboot without coming up correctly it starts disable and delete stuff. Very strange. Maybe there is an option to disable this. Anyhow this is nothing of my changes.

closed 2019-09-17 by stefanbode