stefanbode/Sonoff-Tasmota #185: Shelly 2.5 as roller shutter

zoharaharoni, 2019-09-13:
Hi Stefan,
I'm want to use Shelly 2.5 as roller shutter. I'm very new Home Automation and Tasmota and this will be the first switch I'm adding (Home Assistant). Sorry in advance if the question is very basic!

I installed yesterday the latest bin file from your repository.

I searched for some kind of wiki that explains how to configure shelly 2.5 as roller shutter but couldn't find a complete one so I gathered pieces of information from several places and this is what I came up with:

{"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
SETOPTION80 1
SETOPTION81 1
INTERLOCK ON
INTERLOCK 1,2
SLEEP 0
powerretain 1
rule on energy#current>0.500 do backlog power1 0; power2 0 endon rule 1 rule 5

Resources: 
Your Wiki page - https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-blinds-working-with-two-relays
https://blakadder.github.io/templates/shelly_25.html
This issue: stefanbode#182

I was hoping you could review my configuration and let me know what you think and if it is correct.

Once the basic is OK, I will continue with calibration, etc.

Kind Regards,
Zohar

stefanbode, 2019-09-14:
That’s all fine. I personally do not use sleep 0. I do not see a need. Setoption81 was for a special interlock mode before it got into the main branch of tasmota. This is obsolete. The seems ok.

zoharaharoni, 2019-09-14:
@stefanbode thank you very much! Closing the issue

closed 2019-09-14 by zoharaharoni