stefanbode/Sonoff-Tasmota #186: build for older SDK and older Core version

krachlatte, 2019-09-13:
Hi

the original sonoff version is Core/SDK Version

Program Version | 6.6.0(release-sonoff)
2019-07-06T13:10:20
2_3_0/1.5.3(aec24ac9)

Program Version | 6.6.0.10 stb-1.11(sonoff)
2019-09-08T13:10:06
2_5_2/2.2.2-dev(c0eb301)

is there a way to build 6.6.0.10 for old core/sdk version

meingraham, 2019-09-13:
Follow this procedure: https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod

But use this link to compile Stefan's fork instead.
https://gitpod.io#https://github.com/stefanbode/Sonoff-Tasmota/tree/master

krachlatte, 2019-09-13:
is it possible to put this in the readme wiki ?

thanks for the fast anwer

meingraham, 2019-09-13:
It's in the Tasmota wiki. Not the specific about how to use it for this fork. Hopefully a user can make the small leap between what is in the Tasmota wiki and how to adapt it for Stefan's fork.

krachlatte, 2019-09-13:
building, many thanks

roondar, 2019-09-14:
What is the main purpose ?

meingraham, 2019-09-14:
@roondar 

Of what; Gitpod? It's a web based IDE. In this context, it makes compiling the firmware very simple. Follow the wiki link I sent for more details.

Mike

itanczos, 2019-09-16:
Hello!
My Sonoff Dual (R1) rebooting every 10sec with 2.4 and 2.5 core version. With 2.3 working fine.
Available download bin stefanbode version with old core?

stefanbode, 2019-09-17:
Interesting. I also have some problems with bare esp8266. Maybe I will try the 2.3. anyhow I will not provide all binaries.. too time consuming and there is always someone who want it different. This is the idea of Gitpod to do it on your own

closed 2019-09-17 by stefanbode

meingraham, 2019-09-17:
Give the pre-2.6 Core a try. It is now the default in PlatfromIO for Tasmota. It seems to be a great candidate to finally retire 2.3.