stefanbode/Sonoff-Tasmota #187: NTP not working (precompiled binaries stb-1.11)

electrontube, 2019-09-14:
I have been using the Standard Theo Trends Firmware for the past months since the sub Version didn't work reliably for me.
Today I decided to give Stefans latest Version another try, and flashed one of my devices with one of the precompiled binaries.
It looks like NTP is not working, neither with the default NTP server (pool.ntp.org) nor with any other local or remote NTP server.

Is this a bug, or has NTP been disabled in those precompiled binaries?

stefanbode, 2019-09-15:
No, ntp works reliable on all on my devices. No issues seen so far. Also the deepsleep devices that require ntp work fine. I have found out that tasmota is using ntp not only from the 3 servers you define but also from the router. Maybe a problem there? Keep up posted about the root cause

stefanbode, 2019-09-17:
Did you find the error?

electrontube, 2019-09-17:
It Looks like a strange behaviour of Tasmota, the NTPSERVER1..3 Settings are  ignored when Tasmota thinks the DHCP Server is providing NTP even if it doesn‘t really. I have tried it with a different local DHCP/NTP setup and it works.

closed 2019-09-19 by stefanbode