stefanbode/Sonoff-Tasmota #189: SetOption81 1 - needed with my 3 Button (ex) Tuya "WiFi Curtain Switch"

zwinsch, 2019-09-17:
Hi Stefan,

I am busy modifying (HW+SW) some 3-button Wallswitches with help of your work. Thanks! 
Today I realized, that you did edit the wiki section "Shutter blinds working with two relays" - there you did delete today the hint for "SetOption81 1"
But: in my case I need to set it - otherwise my switches won't work!

I have brand less, called "WF-CS01" (there are others with, the same name but different GPIO Pinout circulating) with three buttons but two relays (with only one opener contact each) (the PCB has space for three, so obvois the same PCB Layout is used for "3-gang" Light switches too).

the 3rd, stop button works for me, if I configure it (and the attached LED) for Relay3 and put it into Interlock 1,2,3 - verry simple, not as in your wiki "How to" - but it works.
But as mentioned: only with Setoption81 1 (as it was written in your Help up to yesterday)

stefanbode, 2019-09-18:
Hmm as far as I remember setoption80 is to enable the shutter and setoption81 was for the special interlock mode. Maybe I messed the two values. Have to take a look at the code

stefanbode, 2019-09-18:
I double checked in the code and setoption81 is not used anymore. Therefor the wiki is correct. Can you explain the problem you had?

zwinsch, 2019-09-19:
Hi Stefan,
it was working like it was ment for self locking - short circuit safe mode: under certain conditions one of the two relays always remained in ON state - after setting Setoption81 1 this behavior changed...
I was using the sonoff.bin  (without -DE) from stb 1.11

stefanbode, 2019-09-19:
Setoption81 is unused. Therefore a correlation is impossible. Must be something else

stefanbode, 2019-09-19:
In failsafe configuration relay2 is setting the direction. This will maintain after a movement. Only relay1 is used to on/off the movement. If you want to relay2 to release you have to add a rule on the event shutter#moved and set the relay to off

closed 2019-10-09 by stefanbode