stefanbode/Sonoff-Tasmota #192: Integrate your shutter code with Tasmota

arendst, 2019-09-22:
Hi Stefan,

you've been working on the shutter code for some time now. I think it must be ready to integrate with Tasmota so you can focus on fixing issues there and do not have to worry any longer to keep up with my changes.

Do I understand correctly that the main code is in xdrv_97_shutter.ino?

meingraham, 2019-09-22:
@stefanbode 

If you would like me to assist, I'm happy to help adding the documentation on this driver to the Tasmota wiki if/once it's merged.

Mike

stefanbode, 2019-09-22:
Hi arend, yes my goal was to stable the code first before starting an integration request. Due to some missing functionality i had to make some minor changes in the main code. For example to get the right source if a relay was changed. Anyhow we should be able to merge it with minor efforts. Let me create a merge request so we can discuss the required changes. Will work on this tomorrow.

arendst, 2019-09-22:
Take all the time you need.

meingraham, 2019-09-22:
@stefanbode Stefan,

With all the work you've done with Tasmota and with how widely your fork is used, it would be great if you would consider joining us on the [Tasmota Discord server](https://discord.gg/Ks2Kzd4). It would be great if you would consider participating in the private Contributors' channel as we bounce off ideas, discuss problems, etc.

If you do, let me know when you've joined so that we can change your role to Contributor so you have access to that channel.

Cheers!

Mike

stefanbode, 2019-09-25:
hi Theo @arendst . should I do the push request against the frequently changing development branch or against the last main branch.

arendst, 2019-09-25:
Pls do development branch.

flashtel, 2019-09-27:
Hi,

Can I reference

https://github.com/stefanbode/Sonoff-Tasmota/issues/188

 merge into Tasmota should resolve.

I'm ready to test when the merge is complete.

Thanks Paul

stefanbode, 2019-09-27:
Hi Theo @arendst. I upgraded my version to todays development branch. Then wipe out everything that is not shutter related and packed the whole branch into a 7z file that is as a splitted file now here in the /bin directory. Sorry I'm to dump to generate a clean pull request. The idea of cloning again and then puting the changes there did not ork wih github. I had to use the clone I already have. Maybe one of the smart guys here can generate a pull request out of the zip file.

arendst, 2019-09-29:
@stefanbode where can I find the 7z file? It's not in the /bin of your master branch.

stefanbode, 2019-09-29:
5 files Sonoff-Tasmota_SHUTTER_MERGE.001 ++ because of limitation of gitub

arendst, 2019-09-29:
Thx.

Got it.

I'll work on integration with current Tasmota.

arendst, 2019-09-30:
@stefanbode Stefan, I've merged your 7z shutter code in Tasmota.

- I refactored it to save code space and increase readability
- I notice that on a wemos test set-up using the web slider only complete open or close is happening; when set to something else it will always go open or close.
- Once the slider is touched on a two relay wemos the relay2 stays always ON
- The code of functions ``ShutterRealToPercentPosition()`` and ``ShutterPercentToRealPosition()`` are a mistery to me; I see no difference in ``set50percent``.
- ``ShutterCalibration`` does nothing other than reporting it's input

Could you be so kind to verify it's functionality and if possible solve the issues please.

I've opened issue https://github.com/arendst/Sonoff-Tasmota/issues/6516 to keep score of progress.

Rgrds, Theo.

stefanbode, 2019-09-30:
I refactored it to save code space and increase readability
- ok
I notice that on a wemos test set-up using the web slider only complete open or close is happening; when set to something else it will always go open or close.
- this is different to my setup. the slider position update send the command shutterposition xx. This should move the shutter to the desired position
Once the slider is touched on a two relay wemos the relay2 stays always ON
- correct. shuttermode=1, you need to set INTERLOCK ON, INTERLOCK 1,2. The is works like you expect. shuttermode=0
The code of functions ShutterRealToPercentPosition() and ShutterPercentToRealPosition() are a mistery to me; I see no difference in set50percent.
- not correct. set50percent is calculation the c1,c2,c3 correction factors for movement. Part of the calibration process. ShutterRealToPercentPosition() and the other calculate position to % and vice versa. Function introduced because c4 and c5 are planned for the nonlinear movement of the shutter
ShutterCalibration does nothing other than reporting it's input
- yes, this is preperation for c4 and c5 calculation. in future you measure cm at 10%, 20% 40% 60% 80% and put in the values. e.g. ShutterCalibration 10.2 23.5 35.5 65 98

stefanbode, 2019-09-30:
//stb mod
  WebGetArg("u1", tmp, sizeof(tmp));
  if (strlen(tmp)) {
    snprintf_P(svalue, sizeof(svalue), PSTR(D_CMND_POSITION "1 %s"), tmp);
    ExecuteWebCommand(svalue, SRC_WEBGUI);
  }

arendst, 2019-09-30:
Start position:

[image]

```
00:00:00 SHT: Accuracy digits: 1
00:00:00 SHT: Shutter 0 (Relay:1): Init. Pos: 20000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 0, c1 200, c2: 200, c3: 0, c4: 0, binmask 3, is inverted 0, shuttermode 0

11:47:19 CMD: interlock
11:47:19 SRC: Serial
11:47:19 MQT: stat/wemos4/RESULT = {"Interlock":"ON","Groups":"1,2"}
```

Set position:
[image]

```
11:47:46 SRC: WebGui from 192.168.2.1
11:47:46 SHT: Position in: payload 48, index 1, source 7
11:47:46 SHT: lastsource 7:, realpos 20000, target 9600, payload 48
11:47:46 SHT: Start shutter in direction -1
11:47:46 SRC: Shutter
11:47:46 SHT: Wait for Motorstop 500
11:47:47 SHT: Shutter 0: Target: 0, powerstatelocal 2
11:47:47 MQT: stat/wemos4/RESULT = {"POWER2":"ON"}
11:47:47 MQT: stat/wemos4/POWER2 = ON
11:47:47 MQT: stat/wemos4/RESULT = {"ShutterPosition2":48}
11:47:47 SHT: Shutter 0: Real Pos: 19300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 0.3  [s]
11:47:48 CFG: Saved to flash at FB, Count 607, Bytes 4096
11:47:48 SHT: Shutter 0: Real Pos: 17300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 1.4  [s]
11:47:49 SHT: Shutter 0: Real Pos: 15300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 2.3  [s]
11:47:50 SHT: Shutter 0: Real Pos: 13300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 3.3  [s]
11:47:51 SHT: Shutter 0: Real Pos: 11200, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 4.4  [s]
11:47:52 SHT: Shutter 0: Real Pos: 9200, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 5.4  [s]
11:47:53 SHT: Shutter 0: Real Pos: 7200, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 6.4  [s]
11:47:54 SHT: Shutter 0: Real Pos: 5300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 7.3  [s]
11:47:55 SHT: Shutter 0: Real Pos: 3300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 8.4  [s]
11:47:56 SHT: Shutter 0: Real Pos: 1300, Target 0, source: Shutter, start-pos: 100 %, direction: -1, rtcshutter: 9.4  [s]
11:47:57 SHT: Shutter 0: Real Pos. 0, Stoppos: 0, relay: 1, direction -1, pulsetimer: 0, rtcshutter: 10.1 [s], operationtime 0
11:47:57 MQT: stat/wemos4/SHUTTER1 = 0
11:47:57 SRC: Shutter
11:47:57 MQT: stat/wemos4/RESULT = {"POWER2":"OFF"}
11:47:57 MQT: stat/wemos4/POWER2 = OFF
11:47:57 MQT: tele/wemos4/RESULT = {"Shutter1":{"Position":0,"direction":0}}
11:47:58 CFG: Saved to flash at FA, Count 608, Bytes 4096
```

ending up in:

[image]

It doesn't seem to stop at 48%

stefanbode, 2019-09-30:
I wonder why it is: 11:47:47 MQT: stat/wemos4/RESULT = {"ShutterPosition2":48}
should be ShutterPosition1. Then it would operate correct

stefanbode, 2019-09-30:
And this is also strange: 11:47:47 SHT: Shutter 0: Target: 0, powerstatelocal 2
Target should here not 0

flashtel, 2019-09-30:
Hi, I've compiled onto DualR2 and 4CHPro. I get the same behavior part slider always goes full open/close. Im still getting the compile error with TasmotaSerial-2.4.0 I revert back to TasmotaSerial-2.3.3 and it's all good ESP 2.5.0 > ESP8285 module. I've got the If/Else code in the compile and it is working. Very good start !!

stefanbode, 2019-09-30:
Agree. Something strange goes on when changing the slider from open to the middle.....

stefanbode, 2019-09-30:
SetOption63 1 has to be mandatory for shutters. I see it came up with both relays ON. Even if INTERLOCK is defined. With option63 ON startup is fine. Now working on the other topic. It has to do of the remove of SRC_IGNORE in one of the checks. It fixes one error and creates another one.

stefanbode, 2019-09-30:
Hi Theo. Is there something I can debug/download/test. Because it works with my current checked in code. Maybe something is missing

stefanbode, 2019-09-30:
Ahh, I see already uploaded. Will check on your code for the issue.

stefanbode, 2019-09-30:
Hi Theo, you missed something in sonoff.ino. The last_source information is really important :-)

```
void SetDevicePower(power_t rpower, uint32_t source)
{
  ShowSource(source);

  //STB mod
  last_source = source;
  //END

  if (POWER_ALL_ALWAYS_ON == Settings.poweronstate) {  // All on and stay on
    power = (1 << devices_present) -1;
    rpower = power;
  }
```

stefanbode, 2019-09-30:
@flashtel : I compiled the current development branch in ATOM IDE against all cores including the shutter code. All worked except the  core_stage.platform. This caused a linker error at the end. 2.4.2 has no errors. Maybe you need to clean up your environment

[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_3_0.platform}
;build_flags               = ${core_2_3_0.build_flags}
;platform                  = ${core_2_4_2.platform}
;build_flags               = ${core_2_4_2.build_flags}
;platform                  = ${core_2_5_2.platform}
;build_flags               = ${core_2_5_2.build_flags}
platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}
;platform                  = ${core_pre.platform}
;build_flags               = ${core_pre.build_flags}

arendst, 2019-09-30:
thx. Will update tomorrow.

arendst, 2019-09-30:
Thx Stefan. Just released and tested OK.

I will change setoption63 to 1 as soon as setoption80 is selected. I guess I also change interlock a soon as one relais of a pair is selected too.

closed 2019-09-30 by arendst

stefanbode, 2019-09-30:
Agree, this is a Good default to start for users. Should I copy my wiki pages into your wiki ? Should we leave setoption80 or move it to a reasonable place. Just take 80 to have enough space not to collide with your additions.

meingraham, 2019-09-30:
@stefanbode 

Yes, please copy the wiki pages and send me the links. I'll incorporate to the appropriate location in the rest of the wiki.

Mike

P.S. Discord invitation still stands :wink: If you have already joined, please send me your ID so we can add you to the Contributor private group. Cheers!

arendst, 2019-10-01:
Yes leave SetOption80 in place. A nice reminder for a long standing request.

Looking at the wiki I noticed the PCF8574 is also a nice feature to have. I'll add it based on your current master branch.

stefanbode, 2019-10-01:
can we postpone the PCF8574. I still have the task to make the INPUT woking. Tested it on the weekend but not successful. Anyhow I agree that the PCF8574 or also MCP23017 is a very nice extension board. But we have to do it in a smart manner.
The last reason why my fork is very famous is the deepsleep option. This is nice for all the plant sensors in the field.

flashtel, 2019-10-01:
Hi,
I did some more side by side testing tonight. There is an issue with the stat message I get a ShutterClose2":0 stat for a shutterclose1 command.

Where should I open this as an issue?

Sonoff-Tasmota 6.6.0.14 by Theo Arends

0:28:50 CMD: shutterclose1
20:28:50 SHT: Start shutter in direction -1
20:28:50 MQT: stat/myplace/house/master_bed1/RESULT = {"POWER2":"ON"}
20:28:50 MQT: stat/myplace/house/master_bed1/POWER2 = ON
20:28:50 MQT: stat/myplace/house/master_bed1/RESULT = {"ShutterClose2":0}

Sonoff-Tasmota 6.6.0.10 stb-1.11 by Theo Arends

20:28:40 Start shutter in direction -1
20:28:40 MQT: stat/myplace/house/master_bed2/RESULT = {"POWER2":"ON"}
20:28:40 MQT: stat/myplace/house/master_bed2/POWER2 = ON
20:28:40 MQT: stat/myplace/house/master_bed2/RESULT = {"shutterclose1":0}

flashtel, 2019-10-01:
Rule Triggers are broken

Sonoff-Tasmota 6.6.0.14 by Theo Arends

21:08:44 CMD: rule1
21:08:44 MQT: stat/myplace/house/master_bed1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":433,"Rules":"ON SHUTTER-1#position==100 DO Publish hm/status/sonoff/level {"%value%"} ENDON"}
21:08:51 CMD: shutteropen
21:08:51 SHT: Start shutter in direction 1
21:08:51 MQT: stat/myplace/house/master_bed1/RESULT = {"POWER1":"ON"}
21:08:51 MQT: stat/myplace/house/master_bed1/POWER1 = ON
21:08:51 MQT: stat/myplace/house/master_bed1/RESULT = {"ShutterOpen1":100}
21:08:52 SHT: Shutter 0: Real Pos: 400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.2  [s]
21:08:53 SHT: Shutter 0: Real Pos: 2400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.2  [s]
21:08:54 SHT: Shutter 0: Real Pos: 4500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.2  [s]
21:08:55 SHT: Shutter 0: Real Pos: 6400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.2  [s]
21:08:56 SHT: Shutter 0: Real Pos: 8500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.2  [s]
21:08:57 SHT: Shutter 0: Real Pos: 10400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.2  [s]
21:08:58 SHT: Shutter 0: Real Pos: 12400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.2  [s]
21:08:59 SHT: Shutter 0: Real Pos: 14400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.2  [s]
21:09:00 SHT: Shutter 0: Real Pos: 16400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 8.2  [s]
21:09:01 SHT: Shutter 0: Real Pos: 18400, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 9.2  [s]
21:09:01 MQT: stat/myplace/house/master_bed1/SHUTTER1 = 100
21:09:01 MQT: stat/myplace/house/master_bed1/RESULT = {"POWER1":"OFF"}
21:09:01 MQT: stat/myplace/house/master_bed1/POWER1 = OFF
21:09:01 MQT: tele/myplace/house/master_bed1/RESULT = {"Shutter1":{"Position":100,"direction":0}}

Sonoff-Tasmota 6.6.0.10 stb-1.11 by Theo Arends

21:08:27 MQT: stat/myplace/house/master_bed2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":433,"Rules":"ON SHUTTER-1#position==100 DO Publish hm/status/sonoff/level {"%value%"} ENDON"}
21:08:36 CMD: shutteropen
21:08:36 Start shutter in direction 1
21:08:36 MQT: stat/myplace/house/master_bed2/RESULT = {"POWER1":"ON"}
21:08:36 MQT: stat/myplace/house/master_bed2/POWER1 = ON
21:08:36 MQT: stat/myplace/house/master_bed2/RESULT = {"shutteropen1":100}
21:08:36 Shutter 0: Real Pos: 900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.5  [s]
21:08:37 Shutter 0: Real Pos: 2900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.5  [s]
21:08:38 Shutter 0: Real Pos: 4900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.5  [s]
21:08:39 Shutter 0: Real Pos: 6900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.5  [s]
21:08:40 Shutter 0: Real Pos: 8900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.5  [s]
21:08:41 Shutter 0: Real Pos: 10900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.5  [s]
21:08:42 Shutter 0: Real Pos: 12900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.5  [s]
21:08:43 Shutter 0: Real Pos: 14900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.5  [s]
21:08:44 Shutter 0: Real Pos: 16900, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 8.5  [s]
21:08:45 Shutter 0: Real Pos: 19000, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 9.5  [s]
21:08:46 MQT: stat/myplace/house/master_bed2/SHUTTER1 = 100
21:08:46 MQT: stat/myplace/house/master_bed2/RESULT = {"POWER1":"OFF"}
21:08:46 MQT: stat/myplace/house/master_bed2/POWER1 = OFF
21:08:46 MQT: tele/myplace/house/master_bed2/RESULT = {"SHUTTER-1":{"position":100, "direction":0}}
21:08:46 RUL: SHUTTER-1#POSITION==100 performs "Publish hm/status/sonoff/level {"100"}"
21:08:46 MQT: hm/status/sonoff/level = {"100"}

arendst, 2019-10-01:
Pls. open it on my repository as there it seems to fail.

Regarding ShutterClose2 I messed something up.
Regarding rules I think something is missed during the implemnation phase.

flashtel, 2019-10-01:
OK, I've just updated the second Dual R2 to Sonoff-Tasmota 6.6.0.14 stb-1.14 by Theo Arends.

I will retest and post the results on your repository.

When you say implementation is that me?? I'm using the same user_config_overide.h for both with the exception of #use_shutter

Thanks

arendst, 2019-10-01:
No it's all me.

flashtel, 2019-10-01:
Your very humble :)  this is an incredible project and I'm very appreciative of your work.