stefanbode/Sonoff-Tasmota #194: Add to PCF8574 input SWITCHES option

rt400, 2019-09-25:
**Have you look for this feature in other issues and in the wiki?**
yes
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**
hi,
i have a 8 relay working with pcf8574 and i want to conncet a 8 touch panel..
can you add the input option?
thanks for the great job..

stefanbode, 2019-09-25:
Oh yes, this is long long on the roadmap. I can also benefit from it to detect the valve end position. I never implemented it because of low urgency. If you have some programmming experience you cn support xdrv_90 is the file where is all comes together. I assume we must use the trigger from the pcf to get a notification that a button was pressed. then sending all states to the defined buttons.

rt400, 2019-09-26:
@stefanbode I wish it was that easy
I also have programming knowledge in C language, but I couldn't understand your code.
I would be happy if you could explain the logic of what you wrote a little, maybe I could see how to combine the buttons.
In my understanding it should be simple, the idea that you created OUTPUT as relays is really great. So in my opinion the INPUT should be automatic as buttons.
The problem I think might be how to associate a button with the relay.?

stefanbode, 2019-10-01:
I'm now working on the INPUT. Still do not understand how they do it on the i2c bus. Interrupt is working but currently, I do not get the states.

stefanbode, 2019-10-09:
Theo has merged the PCF into the master. I tried to get it working but did not succeed. Do not know if my hardware is defect, the code wrong or I do misunderstand the logic. As soon as I get it working I can make a push request to push it into the main project

rt400, 2019-10-09:
@stefanbode i see it and all ready use it with PCF board and relay .
not try with input, where can i found the code for input ?

rt400, 2020-02-05:
@stefanbode any news with INPUT ?

rt400, 2020-04-27:
@stefanbode hi, there is any news with this ?

xopr, 2020-05-16:
Came across this as I was interested in using a PCF8574 combined with Tasmota; the I/O chip is quasi-bidirectional, from what I've understood; setting a port high will cause a strong pull-up within the first SCL cycle (make the rising edge transient; short). After that, a weak pull-up is used (which may depend on the manufacturer; I've seen different values for Texas Instruments and NXP)

One can then read back from the same address (but with the r/w bit high) to see if an external source pulled it low.
I'm not sure how Tasmota would be able to display this since one pin has two features.

stefanbode, 2020-05-18:
I already tried to get this solved but I was not able to read the information back from the PDF. I always got wrong values. To forward this into tasmota is quite easy. I just use SWITCHES. As you can see there is already an input/output selection on the pin. Therefore, prepared. If you know how to o it, please make the changes and contribute. always welcom.

Himbi777, 2020-07-03:
Hello, I have been using 4 pcf8574 on a Wemos D1 for a long time.  However, programmed with Arduino IDE.  It works perfectly.  I am happy to provide the code.

stefanbode, 2020-07-03:
@Himbi777 : This i sgreat can you provide a link

stefanbode, 2020-07-03:
I have tried it with a library that was provided somewhere but was never successful to read the ports in reliable manner

Himbi777, 2020-07-03:
[4x_PCF_8574_Input.txt](https://github.com/stefanbode/Sonoff-Tasmota/files/4869415/4x_PCF_8574_Input.txt)
Hi, here is the code ,I hope the code is useful. I am very interested because I want to switch all of my ESPs to Tasmota.

stefanbode, 2020-07-03:
Dein Version von #include <pcf8574_esp.h> hätte ich auch noch gerne

Himbi777, 2020-07-03:
Sorry, 
[pcf8574_esp.h.txt](https://github.com/stefanbode/Sonoff-Tasmota/files/4869641/pcf8574_esp.h.txt)
Bitte die Endung .txt löschen

xopr, 2020-08-09:
Sorry I couldn't chip in on this having time constraints (there is only so much I can do simultaneously)
I see the library originally comes from https://github.com/WereCatf/PCF8574_ESP where [pcf8574_esp.cpp](https://github.com/WereCatf/PCF8574_ESP/blob/master/pcf8574_esp.cpp) also be found.
So looking at `pcf8574_0` in [4x_PCF_8574_Input.txt](https://github.com/stefanbode/Sonoff-Tasmota/files/4869415/4x_PCF_8574_Input.txt), on line 310, it sets the output values high: you need this because that is your pullup.
After this, if you 'short' a pin to ground, the read
In the loop, it reads the "interrupt" pin and the `read(i)` on lines 430, 434 and 435 give back the pin state (either high, previously set low by software or pulled low by hardware).
Note that it could be made a bit more efficient (it accesses the bus and reads the same byte 24 times); you could call `read8()` and you would have all bits in an 'atomic' call (all bits read at exactly the same point in time).

I archived my test application for an old [Velleman K8000](https://www.velleman.eu/products/view/?id=9383) specifically for two PCF8574A here: https://github.com/xopr/ESPK8000 (Note that it doesn't use a dedicated library apart from `Wire`.
It has a demo mode that does `fillBits()` (or `rollBits()` if you change the code); you can inject arbitrary switch states that are shifted around.

Hope this helps.
If I have more time at hand, I'll try to help, but I guess that won't be before December.

gufonero, 2020-11-28:
Hello,

Is there any command to configure the PCF8574 pins' mode as input/output? I can see that there is a PCF8574 configuration option in web interface, but I couldn't find any reference in the documentation. I use a script sending commands via MQTT to configure my devices.

gorootde, 2021-01-04:
@stefanbode Any updates on this one?

rt400, 2021-01-04:
The best solution I have found is to switch to ESP32, that way there are more inputs and outputs.
Every module I installed with PCF8574 only made me have problems, in the end I found myself with 5 units of PCF8574 that I have nothing to do with ... it's just the result of the experience I have with them.

Himbi777, 2021-01-06:
I have had an esp8266 with 4 pcf8574 in function for over two years.  I programmed the esp with arduino ide.  I need the 32 inputs.  the programming in Arduino works, I sent the corresponding script to Stefan Bode.  he wanted to take it into account if he had the time.  I would prefer a solution with Tasmota too

stefanbode, 2022-05-05:
Implemented in TASMOTA main branch see arendst/TASMOTA

closed 2022-05-05 by stefanbode