stefanbode/Sonoff-Tasmota #195: cannot save new configuration

clh27, 2019-09-28:
Release 6.6.0.14
Installing on a new Shelly 2.5
cannot save WLAN credentials to Flash,
module restarts always with default config.

Using Tasmota stock 6.6.0.14 this isssue does not exist
and config is saved as expected

flashtel, 2019-09-29:
Hi  This is the same behaviour I got on the latest build as well. 

Reference
https://github.com/stefanbode/Sonoff-Tasmota/issues/188

Thanks

jmtalavera, 2019-09-29:
Hi, same here

stefanbode, 2019-09-29:
Yes agree. Also seen here on my side. I have no glue why this happens. Maybe the master branch is defect. I will try to fix and investigate what’s wrong. Please be patient

stefanbode, 2019-09-29:
The error is NOT in the main branch. Therefore something I do prevents from loading the config.....

closed 2019-09-29 by stefanbode