stefanbode/Sonoff-Tasmota #196: Sonoff TX 2-Gang as Roller/shutter switch

deltabert, 2019-09-30:
Hi Stefan,
I am working on a new type Sonoff switch the Sonoff TX 2-Gang for use with screens.
From the page of Theo Arends i found that this new Sonoff TX series switches work the same as the old Sonoff T1 series switches. 
So i loaded your firmware and selected the Template based on Sonoff T1 2CH   (29).
I used firmware version "Sonoff-Tasmota 6.6.0.10 stb-1.11" because this was loaded OK. (With newer versions i had problems with the WiFi configuration after loading. Besides my Teepao switches are working OK with this version.)

Without setting the options for the the interlock for both relays, the relays are spontaniously switching ON and OFF. This applies for both relays. 
`13:14:07 MQT: tele/Sonoff_2TX/STATE = {"Time":"2019-09-30T13:14:07","Epoch":1569845647,"Uptime":"0T00:15:18","UptimeSec":918,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"NoAccess-3","BSSId":"00:12:17:07:A5:DD","Channel":6,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:08","DeepSleep":0,"Heap":29216}}
13:14:14 MQT: stat/Sonoff_2TX/RESULT = {"POWER2":"ON"}
13:14:14 MQT: stat/Sonoff_2TX/POWER2 = ON
13:14:18 MQT: stat/Sonoff_2TX/RESULT = {"POWER2":"OFF"}
13:14:18 MQT: stat/Sonoff_2TX/POWER2 = OFF
13:14:20 MQT: stat/Sonoff_2TX/RESULT = {"POWER1":"ON"}
13:14:20 MQT: stat/Sonoff_2TX/POWER1 = ON
13:15:13 MQT: stat/Sonoff_2TX/RESULT = {"POWER1":"OFF"}
13:15:13 MQT: stat/Sonoff_2TX/POWER1 = OFF
13:16:46 MQT: stat/Sonoff_2TX/RESULT = {"POWER1":"ON"}
13:16:46 MQT: stat/Sonoff_2TX/POWER1 = ON
13:17:08 MQT: stat/Sonoff_2TX/RESULT = {"POWER1":"OFF"}
13:17:08 MQT: stat/Sonoff_2TX/POWER1 = OFF
13:18:25 MQT: stat/Sonoff_2TX/RESULT = {"POWER2":"ON"}
13:18:25 MQT: stat/Sonoff_2TX/POWER2 = ON
13:19:07 MQT: tele/Sonoff_2TX/STATE = {"Time":"2019-09-30T13:19:07","Epoch":1569845947,"Uptime":"0T00:20:18","UptimeSec":1218,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"NoAccess-3","BSSId":"00:12:17:07:A5:DD","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08","DeepSleep":0,"Heap":29216}}
13:19:33 MQT: stat/Sonoff_2TX/RESULT = {"POWER2":"OFF"}
13:19:33 MQT: stat/Sonoff_2TX/POWER2 = OFF`

And after i set the interlock options, this remains the same...
Any idea what is going wrong here?
Regards, Bert

stefanbode, 2019-09-30:
if you set weblog 5 or seriallog5 you will see "who" is switching your relays.

deltabert, 2019-09-30:
Hi Stefan, thank you for your reaction.
This is the output of "weblog 5":
`13:40:46 CMD: weblog 5
13:40:46 MQT: stat/Sonoff_1TX/RESULT = {"WebLog":5}
13:40:46 CFG: Saved to flash at F9, Count 91, Bytes 4100
13:40:54 WIF: Checking connection...
13:40:54 WIF: Connected
13:41:14 WIF: Checking connection...
13:41:14 WIF: Connected
13:41:22 APP: Button2 multi-press 1
13:41:32 SRC: Button
13:41:32 Switched relay: 2 by Button
13:41:32 MQT: stat/Sonoff_1TX/RESULT = {"POWER2":"ON"}
13:41:32 MQT: stat/Sonoff_1TX/POWER2 = ON
13:41:32 CFG: Saved to flash at F8, Count 92, Bytes 4100
13:41:33 APP: Button1 multi-press 1
13:41:34 WIF: Checking connection...
13:41:34 WIF: Connected
13:41:43 Switched relay: 2 by I
13:41:43 MQT: stat/Sonoff_1TX/RESULT = {"POWER2":"OFF"}
13:41:43 MQT: stat/Sonoff_1TX/POWER2 = OFF
13:41:43 SRC: Button
13:41:43 Switched relay: 1 by Button
13:41:43 MQT: stat/Sonoff_1TX/RESULT = {"POWER1":"ON"}
13:41:43 MQT: stat/Sonoff_1TX/POWER1 = ON
13:41:43 CFG: Saved to flash at F7, Count 93, Bytes 4100
13:41:51 APP: Button1 multi-press 1
13:41:54 WIF: Checking connection...
13:41:54 WIF: Connected
13:42:00 SRC: Button
13:42:00 Switched relay: 1 by Button
13:42:00 MQT: stat/Sonoff_1TX/RESULT = {"POWER1":"OFF"}
13:42:00 MQT: stat/Sonoff_1TX/POWER1 = OFF
13:42:01 CFG: Saved to flash at F6, Count 94, Bytes 4100`

It looks as if the relay is activated by the button... But the Switch is laying on my desk and nothing is touching it. It is laying on a paper document and i do not think that static electricity could be involved. I have about 12 Sonoff switches (No touch) and 2 Teepao switches and i am always working with them this way...
I do not understand this. Hope you have some ideas?

stefanbode, 2019-09-30:
can you connect you button to ground. it look like there is no pullup (10k Ohm) resistor included keeping the button open. open pins are most times not a good idea.

deltabert, 2019-10-02:
I think the button contacts are only correctly terminated when the device connected to mains.
I had the device assembled (logic board and relay part) but device not connected to mains but supplied the 3.7 V via the programming connections and the FTD1232.
That worked for all my Sonoff and also Teepao devices so far...  But it seems to give problems with this Sonoff TX.  It looks like it is working correctly now. Lest see for a couple of days.

Thank you so far for thinking with me.

closed 2019-10-02 by deltabert