stefanbode/Sonoff-Tasmota #197: Sonoff 4CH Pro won´t save wlan credentials?!

sledgeha1, 2019-10-03:
Hi,

i just flashed your Tasmota fork on a fresh Sonoff 4CH Pro R2 device. I did use the pre compiled bins in your code/bin/sonoff.bin folder. 

After flashing i can connect to the hotspot and enter my wlan credentials. But the device always comes back to open a hotspot and ask for wlan settings. It seems as it is not possible to save something to flash.

I then tried the latest original tasmota bin. With that bin flashed to device i can put in my wlan credentials an the device shows up in my network after reboot.

The last thing i tried to do was flashing a minimal image over tasmotas web interface and then flash your bin  over the interface.

Doing that the device flashed with your fork shows up in my network. But again i´m not able to save any setting like mqtt, module etc.

ranseyer, 2019-10-04:
This works for me, with the 2.4.2 or 2.3 Core on the Sonoff 4CH Pro...
(Flashmode: DOUT /  yes, wipe all Data) 
http://thehackbox.org/tasmota/sonoff.bin

sledgeha1, 2019-10-04:
Thanks for your Feedback. I just compiled from sources and got it working.

stefanbode, 2019-10-05:
Yes I have to update the bin files. I had an issue with the config file. 4 bytes to long. Therefore is did not accept it. This is fixed in the current version. Saving config should generate 4096 bytes file.

closed 2019-10-05 by stefanbode