stefanbode/Sonoff-Tasmota #198: SetOption80 1 HangUp the Device

onkeldittmayer, 2019-10-05:
Is it possible, that since the integration of the Shuttercode to arendst Sonoff the SetOption80 doesn't work anymore?
When i compile the blank Stefanbode-fork and set the Option80 to 1 the hole Device stops responding...
i compiled three times (success) with Atom.

flashtel, 2019-10-05:
I've tested on both the merged code and the latest Stephan code and it is working. I've been running it for a few days now on shutters running many times a day. It has been stable. I've compiled with both Atom and Arduino IDE.

flashtel, 2019-10-05:
The only issue is the stat messages are different between the two. shutteropen1 (Theo) vs ShutterOpen1 (Stephan) and them there is a difference in the tele messages if your using rules  shutter1 (Theo) vs Shutter-1 (Stephan)

If you want to use the main code, you need define USE_SHUTTER in user_config_override.h

In platformio.ini I make these mods

board                     = esp8285

platform                  = ${core_2_5_2.platform}
build_flags               = ${core_2_5_2.build_flags}

default_envs = sonoff

flashtel, 2019-10-05:
backlog SETOPTION80 1; SETOPTION63 1; interlock on; interlock 1,2,3,4; SHUTTERRELAY2 3; topic house/bathroom;

backlog SHUTTEROPENDURATION1 32; SHUTTERCLOSEDURATION1 32; SHUTTEROPENDURATION2 14;  SHUTTERCLOSEDURATION2 14;

bjeram, 2019-10-05:
I also have a problem after I set: SetOption80 the Sonoff Dual R2 starts to reboot. From the serial log message looks like after trying to apply the configuration. It might be that there is some leftovers configuration from the previous installation. I need to investigate a bit more: erase the whole flash before flashing, give a try to original Tasamoaot, ....

onkeldittmayer, 2019-10-05:
How many days does it run? This fork was updated 3 days ago....
i just try it once again. set Setoption80 to 1 doesn't work for me.
After a restart Tasmota doesn't come back.
Curious: i find a W-Lan network: ESP-100D1D but can't connect... very strange.

The Theo develop Code work with shutter and option80 1
so first i'll play with these.

bjeram, 2019-10-05:
I tried with the version that was updated 3 days ago. What do you see on the serial output? 
BTW: what HW do you use?  How do you build (plafromio?) ?

meingraham, 2019-10-05:
@onkeldittmayer 

Yes, this feature was just recently ported to the main branch. If it says 3 days, then yes, that's right; this week.

Did you compile your own binary?

`ESP-100D1D` usually indicates corrupted flash contents. You'll need to erase_flash and upload Tasmota again.

stefanbode, 2019-10-06:
I’m off this weekend. Therefore cannot compile and recheck again. Anyhow setoption80 did not change at all. Therefore it must work. Can give more details Monday afternoon

onkeldittmayer, 2019-10-06:
@bjeram have to check the serial output. i build with Atom and platformio. I flash the Moeshouse 2 way 2 gang MS-104B Relay via an Arduino UNO (with no IC on it).
@meingraham yes, but i also build the raw fresh bode code and got the same issue.

like i said: Theos code work... let me check serial out...

bjeram, 2019-10-06:
> after setting setoption80 to 1 and restart
> the following seems to repeat over and over again:
> 
> ets Jan 8 2013,rst cause:1, boot mode:(3,7)
> 
......
> 00:00:00 CFG: aus Flash geladen am F8, zählen 12
> 00:00:00 SHT: Accuracy digits: 0

I can see the same, so it looks like it is a problem to  load/apply configuration.

ihavenonick, 2019-10-06:
Ich kann es bestätigen, sobald Deine Fork größer 6.6.0.10 mit Core 2.5.2 (nur damit kann ich ein 2MB bin kompilieren das auch 2MB Flash verwendet) verwendet wird, hängt sich ein Shelly 2 oder 2.5 komplett auf und muss erst gelöscht werden damit man ihn wieder erreicht.

stefanbode, 2019-10-07:
I can see that: `00:00:00 SHT: Accuracy digits: 0 `is reported. This should be `00:00:00 SHT: Accuracy digits: 1`. This assumes there is strange data in the configuration. Please do a "`reset 2`" to ensure everything is back to default. At least after the first reboot there is no "0" allowed.

stefanbode, 2019-10-08:
There is a fix coming in a minute. DIV0 exception is fixed. Also, update the BIN files

closed 2019-10-09 by stefanbode