stefanbode/Sonoff-Tasmota #201: SHUTTER50PERCENT not work

carlosribeiro, 2019-10-09:
in version 6.6.0.10 the SHUTTER50PERCENT command does not work. gives the result of the console "unknown"

stefanbode, 2019-10-09:
yes, Theo has renamed in TASMOTA Master the command to shuttersethalfway. I will update the wiki. Please take a look at the new calibration option, if intrested. Also documented in the wiki

snakuzzo, 2019-10-09:
Hi @stefanbode....rollershutter configuration wiki was deleted ? I can't find the guide...the old page was removed ?

stefanbode, 2019-10-09:
Yes this is now part of Tasmota. The wiki is here https://github.com/arendst/Sonoff-Tasmota/wiki/Blinds-and-Roller-Shades

snakuzzo, 2019-10-09:
Great!!! :)
Thank you

snakuzzo, 2019-10-09:
@stefanbode I missed this update...when this function was included in Tasmota?

jokerigno, 2019-12-21:
Hi stefanbode, what it the best way to move from your fork to official tasmota? 
I mean I think that If I reflash my sonoffr2 I will loose al my config, including shutter time and percentage for 50% so I will have to restart from scratch for every blind. Am I right?

Or maybe there's some tip that you can share with use helping migrating? 

Thank you in advance

stefanbode, 2019-12-21:
Depend on. If you use one of the later releases you will not loose anything. Just give it a try. If you loose I am very sorry. You have to go through this one time

closed 2019-12-22 by stefanbode