stefanbode/Sonoff-Tasmota #203: Hello,

eisi123, 2019-10-19:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

eisi123, 2019-10-19:
Hello, it is possible to dim the lights in the switch?

stefanbode, 2019-10-20:
No, a Switch has just on and off. There are PwM Controller that might dim a led, if the LeD is suitable for this. There are also LED devices from sonoff. The relays just can on and off. Maybe you also take a look for devices from shelly. They also have a dimmer.

eisi123, 2019-10-20:
Oh, I am sorry. I think, we misunderstood. :-)
In the shutter switch are three leds. Up/Pause/Down. These leds are shinging very bright.
My question was, if these three leds are dimable by the firmware.

stefanbode, 2019-10-20:
WhAt device are you using?

eisi123, 2019-10-20:
Teepao 3rd Generation

stefanbode, 2019-10-21:
Hmm that’s not an easy answer without disassemble the device and take a look how they attached the LeD of the Switches. Sorry I cannot help. I assume the device has an ESP8266.

closed 2019-10-21 by stefanbode