stefanbode/Sonoff-Tasmota #204: Shutter dead time support

to-scho, 2019-10-25:
Hi Stefan, thanks for introducing shutter support in Tasmota. Unfortunately it seems that my Somphy Motors have significant dead time when starting or stopping motors. I am running Short Circuit Safe mode.Can you add some extra ms setting for any movement in addition to open close duration? In addition would it make sense to have some extra ms when shutter position is set to 0 or 100? With that any integrated timing error will at least ‘recalibrate’ when moving to any end position. Regards, Tobias

stefanbode, 2019-10-26:
Hi Tobias, the extra seconds at the end are already included. Can you make the following movement and and print the cm - Position here. this make it easier for me see what is wrong.

shutterposition 20
shutterposition 40
shutterposition 60
shutterposition 80
shutterposition 20
shutterposition 80
shutterposition 20

to-scho, 2019-10-27:
Hi, 
here we go:

shutteropenduration 20.8
shuttercloseduration 18.5

shutterposition 100 135cm
shutterposition 0 0cm
shutterposition 20 5cm
shutterposition 40 25cm
shutterposition 60 47cm
shutterposition 80 72cm
shutterposition 20 ~0cm, einige Lamellen noch offen
shutterposition 80 72cm
shutterposition 20 ~0cm, einige Lamellen noch offen

As you can see main problem is that I cannot open the shutter to shutterposition 100 once stopped in between. From shutterposition 80=72cm i'ts far too long to reach the 135cm with 20% movement.

Da geht irgendwas schief ;-)

Regards, Tobias

to-scho, 2019-10-27:
Sorry, the mail from GitHub truncated the last line with your question to the calibrationparameters:
 
SHT: Shutter 0 (Relay:1): Init. Pos: 41600 [100 %], Open Vel.: 100 Close Vel.: 112 , Max Way: 41600, Opentime 20.8 [s], Closetime 18.5 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1

stefanbode, 2019-10-27:
Uhh, this is quite a lot of movement you lose. If you understand right normally 100 = 135cm and if you do it stepwise you with roughly end at 100cm; roundabout. There are 5 starts. Your opentime is 20.8. That does mean: 135/20.8 = 6.5cm/sec. If we lose 35cm in 5 starts this is 7cm per start wich is roughly 1 second that the motor needs from the start command to effectively go to start. Wow. Let me check If I can compensate for this. I assume it is the same for up and down. My idea is to change the startposition internally to "bring in" the additional runtime. I need to ensure the position at the end and on a position during movement is as precise as it can be.

to-scho, 2019-10-27:
Hi Stefan,
after experiment today I am not quite sure whether I simply loose some seconds. If you look to the last positions 20/80/20 I got quite accurate results when toggling between 20 and 80. Anything else I might have missed? My durations are calculated from Speed/cm and the 135. When moving 100% Everything is fine and I can here the Motor stops shortly before the Relais opens. 
Tobias

stefanbode, 2019-10-27:
Hi Tobi, that 20/80/20/80 will work fine is correct. This was just my double check, that you really have a start problem. if you do 20/40/60/80/20/40/60/80/20 you will see it get worse. I have an idea where to hook in. just struggeling with the first move them position is 0. then my start is negative and this is what the software currently does not like....

stefanbode, 2019-10-27:
Hi Tobi, I just uploaded version 1.17. Maybe you can give it a try. If you start you will get a new config line with "motordelay" at the end.

`00:00:00 SHT: Shutter 0 (Relay:1): Init. Pos: 20000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoedffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 0`

This delay are multiple of 0.05sec. e.g. 20 is 1 sec (20x0.05). To set the motordelay please use the new command and put in the time in SECONDS!!! For your problem please start with:

`shuttermotordelay 1.1`

The init line should answer with "motordelay 22"

stefanbode, 2019-10-27:
Please let me know if this solves you problem. if so, I will update the master project with the new code. As you might know the shutter is now also in the MAIN TASMOTA project included.

to-scho, 2019-10-27:
Hi Stefan,

thx for the update. I am already running main tasmota project with shutters. So I took your changes on top of 6.7.1.

I ended up with shuttermotordelay 0.9 for my setup. Afterwards shutter ends at same position before and after 20/40/60/80/20 sequence.

Please go ahead and update tasmota main repo.

Regards, Tobias

stefanbode, 2019-10-27:
Ok great to hear. Will submit a change request against the main project tomorrow.

stefanbode, 2019-10-30:
Change is submitted to the master. Wiki is updated to describe the procedure.

closed 2019-10-30 by stefanbode