stefanbode/Sonoff-Tasmota #205: Vcc in state

Gisbert1, 2019-10-28:
Dear Stefan,

in arendst Tasmota branch, there is in State the Vcc of the device:

> 12:39:48 MQT: tele/Lux.Ostseite/STATE = {"Time":"1970-01-11T12:39:48","Uptime":"10T12:03:33",**"Vcc":3.177**,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":4,"Wifi":{"AP":1,"SSId":"IoTXXXXX","BSSId":"FE:EC:DA:3E:A8:C4","Channel":3,"RSSI":42,"LinkCount":1,"Downtime":"0T00:03:23"}}

How can I get the Vcc reading in your branch?

Best Regards
Gisbert

meingraham, 2019-10-28:
#define USE_ADC_VCC                              // Display Vcc in Power status.

This is disabled by default so that ADC0 can be used as a user input. If you enable internal Vcc, the ADC0 input cannot be used.

Gisbert1, 2019-10-28:
Found it here: [https://github.com/stefanbode/Sonoff-Tasmota/blob/master/sonoff/my_user_config.h](url)
I need to compile the firmware, but I do not know how to do this. I'm using Arduino.
Can you guide me?

Best Regards
Gisbert

meingraham, 2019-10-28:
Oh, yikes. You haven't looked at any of the Tasmota documentation?

Start here - https://github.com/arendst/Tasmota/wiki/Prerequisites

Use Gitpod to compile.

closed 2021-12-08 by stefanbode