stefanbode/Sonoff-Tasmota #206: Alexa Dimming doesnt work

eduardocanton, 2019-11-06:
hello,

I have the last version installed over a sonoff dual r2. I have configured it to use with a blind:
(with "interlock 1", "SetOption80 1" and "SetOption81 1"). 

the device works fine, with a slide to change the blind status.

The problem is with alexa emulator. I active it with "hue bridge", but alexa discover it like a bulb without dimming and then i cannt use " alexa , open blind to 50%", only open and close.

Must I do something more to get it to work ?

stefanbode, 2019-11-06:
No, there is no shutter in alexa because I use the hue interface. This uses the blind as a dimmer. not straight forward but ok. I haven't used the alexa for a long time. it should be a dimmer and then "blind to 50%" should work. Anyhow open close does not work. only on/off. As mentioned it is a dimmer.

eduardocanton, 2019-11-06:
yeah, I mean on / off  action with alexa. If I access for example with alexa app , inside the devices. this devices doesnt have the slide to set the dimmer. I tested with the original tasmota and if it´s works... then, i dont know what it ¡s different

closed 2021-12-13 by github-actions[bot]