stefanbode/Sonoff-Tasmota #207: PCF8574 on master

betyar, 2019-11-15:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP-01
- [X] Tasmota binary firmware version number used: 6.7.1-sonoff
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esptool.py 
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

**(Please, remember to close the issue when the problem has been addressed)**

Previously when I used the forked version (6.6) everything worked "out of the box". When I read that the fork was merged with the master I flashed my ESP-01 with version 6.7.1. The PC8574 is found when running i2cscan, but unlike the original fork a separate config page doesn't appear. The Tasmota wiki also doesn't indicate which bin version needs to be flashed (sonoff.bin? sonoff-sensors.bin?). I know that with the MCP230xx GPIO Expander a Sensor29 command has to be used for configuration. Is something similar used on the master or should there be the config page that was on the fork? In other words, can the PC8574 be used with the master or should I go back to using the fork?

stefanbode, 2019-11-15:
you can use the MASTER TASMOTA. Anyhow you need to enable the driver in your user_config_override and compile it on your own. There is no precompiled version

betyar, 2019-11-16:
Thanks. I will stick with the forked version for now because I would like to keep up to date with the latest developments. It would be great when the input issue is resolved. I am also interested in your work on shutters and blinds, so by continuing to use the fork I have a reason to check back frequently. :)

Best regards and keep up the good work!

closed 2019-11-16 by betyar