stefanbode/Sonoff-Tasmota #208: low voltage on PCF8574

betyar, 2019-11-17:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

I am having trouble using a 5V relay module with the PCF8547. It appears there isn't enough voltage to drive the relay, I am measuring only 1.7 volts. The LED indicator on the module is dim showing that the relay has been triggered but there isn't enough voltage to saturate the coil. The same issue occurs if I use a simple LED instead of a relay, it's very dim because of low voltage. For the LED I am using a 220R resistor, the relay trigger is connected directly to the PCF8547 (the relay board itself is powered by a separate 5V connection from the same power source). This setup works no problem when connected to a Wemos GPIO directly. 

Another problem is when the system boots the PCF8547 port turns on automatically even though the web GUI indicates that it is off. When I toggle it on and then off again everything is fine, the port is off and is synced with the web GUI. Inverting the ports don't help because they will never be in sync with the web GUI and I don't want the relay to start in a NC state.

Any ideas how to increase the voltage to the PCF8547 ports and to have them not turn on when starting?

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos Mini ESP-12F
- [X] Tasmota binary firmware version number used: 6.6.0.16 stb-1.16(sonoff)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esptool.py
- [X] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
17:11:55 CMD: Backlog Template; Module; GPIO
17:11:55 MQT: stat/wemos/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:11:55 MQT: stat/wemos/RESULT = {"Module":{"0":"Generic"}}
17:11:55 MQT: stat/wemos/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [X] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

17:14:07 MQT: stat/wemos/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"wemos","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:14:07 MQT: stat/wemos/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:10:10","StartupUTC":"2019-11-17T16:03:57","Sleep":50,"CfgHolder":4617,"BootCount":51,"SaveCount":162,"SaveAddress":"F8000"}}
17:14:07 MQT: stat/wemos/STATUS2 = {"StatusFWR":{"Version":"6.6.0.16 stb-1.16(sonoff)","BuildDateTime":"2019-10-09T07:26:57","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
17:14:07 MQT: stat/wemos/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["betyar0",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
17:14:07 MQT: stat/wemos/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000381"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,27,28","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:14:07 MQT: stat/wemos/STATUS5 = {"StatusNET":{"Hostname":"wemos-3959","IPAddress":"192.168.1.115","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"2C:F4:32:20:6F:77","Webserver":2,"WifiConfig":4}}
17:14:07 MQT: stat/wemos/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.104","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_206F77","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:14:07 MQT: stat/wemos/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 17 16:14:07 2019","Local":"Sun Nov 17 17:14:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:02","Sunset":"17:08"}}
17:14:07 MQT: stat/wemos/STATUS10 = {"StatusSNS":{"Time":"2019-11-17T17:14:07"}}
17:14:07 MQT: stat/wemos/STATUS11 = {"StatusSTS":{"Time":"2019-11-17T17:14:07","Uptime":"0T00:10:10","UptimeSec":610,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":24168}}}
17:14:11 MQT: tele/wemos/STATE = {"Time":"2019-11-17T17:14:11","Uptime":"0T00:10:14","UptimeSec":614,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":27504}}

[... 7 lines cut ...]
00:00:00 PCF: Total devices 1, PCF8574 output ports 1
00:00:00 Project sonoff Sonoff Version 6.6.0.16 stb-1.16(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 betyar0 in mode 11N as wemos-3959...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on wemos-3959 with IP address 192.168.1.115
17:04:03 MQT: Attempting connection...
17:04:03 MQT: Connected
17:04:03 MQT: tele/wemos/LWT = Online (retained)
17:04:03 MQT: cmnd/wemos/POWER = 
17:04:03 MQT: tele/wemos/INFO1 = {"Module":"Generic","Version":"6.6.0.16 stb-1.16(sonoff)","FallbackTopic":"cmnd/DVES_206F77_fb/","GroupTopic":"sonoffs"}
17:04:03 MQT: tele/wemos/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos-3959","IPAddress":"192.168.1.115"}
[... cut ...] "betyar0","BSSId":"00:23:CD:11:B1:9C","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04","DeepSleep":0,"Heap":28096}}
17:05:24 CMD: cmnd/wemos/POWER
17:05:24 MQT: stat/wemos/RESULT = {"POWER":"OFF"}
17:05:24 MQT: stat/wemos/POWER = OFF
17:05:53 MQT: stat/wemos/RESULT = {"POWER":"ON"}
17:05:53 MQT: stat/wemos/POWER = ON
17:05:54 MQT: stat/wemos/RESULT = {"POWER":"OFF"}
17:05:54 MQT: stat/wemos/POWER = OFF
 ```

**(Please, remember to close the issue when the problem has been addressed)**

betyar, 2019-11-19:
Update:

The first problem concerning low voltage has been more or less resolved. When I measured 1.7 volts it was with the LED and a 220R resistor attached. Without the LED I got 2.45 volts, so I simply used a smaller resistor than what I would usually use. Meanwhile, I solved the problem with the relay by using a MOFSET between the PCF8547 port and the module's trigger pin.

The second issue unfortunately hasn't been resolved yet. The PCF8547 ports are on at startup while the GUI indicates that it's off, so I still need to turn the port on and off at the beginning. I probably can write a rule to do this, but I don't like the idea of having the relays and LEDs quickly switch on and then off when starting or rebooting the device.

stefanbode, 2019-11-21:
I will try on the weekend to reproduce this. I do not see the issue with the release I loaded into my watering equipment

betyar, 2019-11-21:
Thanks. By the way, here is the workaround I am currently using. It's a rule which switches all ports on and then off again:

> on Power1#Boot do backlog power0 on; RuleTimer1 1 endon  on Rules#Timer=1 do power0 off endon

When the system starts the ports are on for about three seconds and then switch off. The web GUI is also subsequently synced as a result. 

Since I will be using the PCF8574 on a board with 4 relays connected to lights which can be turned on not only by WiFi but also RF433 as well as a manual switch, it's not that bad because hopefully it doesn't have to be powered on or reset often. Still, when it does (such as in the case of a power failure) it's nevertheless a little strange to have all four sets of lights turn on for a few seconds at the beginning.

yogesh102011, 2020-01-23:
if i make all 8 as output..its working fine with relay. but one problem i am facing is if all the 8 relays are on.. while restarting none of the relay is in onstate. but in web interface it show all are on. if i off one relay all are getting on except the one which is off

stefanbode, 2020-01-23:
Ok, this looks like the current state is not send on the init process. I will fix this in the master tasmota and propagate here. Please change to the original tasmota, if possible because nearly all features are now in the main project

yogesh102011, 2020-01-23:
what i have to do now.. what means original tasmota.. if seven relays are on. if i restart no issue all are syns well. if all 8 relays are on..on restart problem arises.. none of the relay will get on. in web interface all the relays will be in on state.  if i tried changing any one to off in web ui..it will on the remaining seven relays.

yogesh102011, 2020-01-23:
i compiled from development version 

Program Version | 8.1.0.4(tasmota)
-- | --
2020-01-23T00:49:07
2_6_1/2.2.2-dev(38a443e)