stefanbode/Sonoff-Tasmota #211: MQT: TLS connection error: 0

Hadea4711, 2019-11-26:
Hi,

I installed the latest version of Tasmota ("Tasmota 7.0.0.3 stb-1.19 von Theo Arends") for my Shelly 2.5.

I set up the MQTT with the same credentials, I've tested with MQTTfx.

MQTTfx is connecting to the broker inmediately BUT not my Tasmota :-(.

In the Console is written:
MQT: TLS connection error: 0
MQT: connection failed due to 192.168.0.100:1883, rc -2. retry in 10 s

I didn't find anything with this error message. Tried the same with my sonoff ("Sonoff-Tasmota 6.7.1 von Theo Arends") which connect directly with the MQTT broker.

Is there something wrong in the bin?

KR
Sebastian

MacPolak, 2019-11-28:
i have exactly the same issue, lot of devices on 6.7.1 works great just one Tasmota 7.0.0.3 stb-1.19 can't connect. The same message from Tasmota. Message from the broker: 
1574967786: New connection from 192.168.0.130 on port 1883.
1574967786: Socket error on client <unknown>, disconnecting.
Pls help

Hadea4711, 2019-11-28:
I have compiled a new Version of Tasmota (original) on the 6.7.1 basis. 

Now it works. In my point of view there must be something wrong in the file, we are using.

MacPolak, 2019-11-29:
I think I will do the same thing, what did you add in the config fot Tasmota to work as a roller shutter?

Hadea4711, 2019-11-29:
Here the link to the "[Online Compiler](https://gitpod.io/#https://github.com/benzino77/tasmocompiler)".

You have to add the following code in the **user_config_override.h** to activate the shutter support in Tasmota:

#ifndef USE_SHUTTER
#define USE_SHUTTER           // Add Shutter support (+6k code)
#endif

Please also read this topic in Github: [Compile your build](https://github.com/arendst/Tasmota/wiki/Compile-your-build)

closed 2019-11-29 by Hadea4711