stefanbode/Sonoff-Tasmota #216: Shutter with LC-TECH X2 in mode 1 allows changing direction without power removal

taste66, 2020-02-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am using a LC-Tech X2 relay to control my shutter. This relay needs rules to change the relay status over the serial link to the extra processor. (see https://github.com/arendst/Tasmota/wiki/LC-Technology-WiFi-Relay) 
If you change the shutter direction while moving the commands to stop and start the motor ar to close so do not propagate to the actual relais. This breaks the direction relay if this action is performed while moving.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): LC-Tech X2
- [ ] Tasmota binary firmware version number used: _8.0.1.7
  - [ ] Pre-compiled
  - [ X] Self-compiled
    - [X ] IDE / Compiler used: platformIO_____
- [ X] Flashing tools used: Tasmotizer_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
20:25:36 CMD: Backlog Template; Module; GPIO 255
20:25:36 MQT: stat/Shutter/01/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:25:37 MQT: stat/Shutter/01/RESULT = {"Module":{"18":"Generic"}}
20:25:37 MQT: stat/Shutter/01/RESULT = {"GPIO0":{"21":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"22":"Relay2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:26:23 MQT: stat/Shutter/01/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":253,"Rules":"on System#Boot do Backlog Baudrate 115200; SerialSend5 0 endon  on Power1#State=1 do SerialSend5 A00101A2 endon  on Power1#State=0 do SerialSend5 A00100A1 endon  on Power2#State=1 do SerialSend5 A00201A3 endon  on Power2#State=0 do SerialSend5 A00200A2 endon"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:27:47 MQT: stat/Shutter/01/STATUS = {"Status":{"Module":18,"FriendlyName":["Shutter_01","Shutter_01a"],"Topic":"Shutter/01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
20:33:02 SHT: Shutter1 Real 29605, Start 40000, Stop 0, Dir -1, Delay 0, Rtc 4.95 [s], Freq 1052
20:33:02 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":74,"Direction":-1}}
20:33:03 SHT: Shutter1 Real 27505, Start 40000, Stop 0, Dir -1, Delay 0, Rtc 5.95 [s], Freq 1052
20:33:03 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":69,"Direction":-1}}
20:33:03 SHT: Wait for Motorstop..
20:33:03 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-13T20:33:03","Uptime":"0T00:37:06","UptimeSec":2226,"Vcc":3.432,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:33:03 MQT: stat/Shutter/01/RESULT = {"POWER1":"OFF"}
20:33:03 MQT: stat/Shutter/01/POWER1 = OFF (retained)
20:33:04 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-13T20:33:04","Uptime":"0T00:37:07","UptimeSec":2227,"Vcc":3.434,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:33:04 MQT: stat/Shutter/01/RESULT = {"POWER1":"OFF"}
20:33:04 MQT: stat/Shutter/01/POWER1 = OFF (retained)
20:33:04 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-13T20:33:04","Uptime":"0T00:37:07","UptimeSec":2227,"Vcc":3.432,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:33:04 MQT: stat/Shutter/01/RESULT = {"POWER2":"OFF"}
20:33:04 MQT: stat/Shutter/01/POWER2 = OFF (retained)
20:33:04 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-13T20:33:04","Uptime":"0T00:37:07","UptimeSec":2227,"Vcc":3.434,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:09"}}
20:33:04 MQT: stat/Shutter/01/RESULT = {"POWER1":"ON"}
20:33:04 MQT: stat/Shutter/01/POWER1 = ON (retained)
20:33:04 MQT: stat/Shutter/01/RESULT = {"ShutterOpen1":100}
20:33:04 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
20:33:04 MQT: stat/Shutter/01/RESULT = {"SerialSend":"Done"}
20:33:04 SHT: Shutter1 Real 27395, Start 27295, Stop 40000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 1000
20:33:04 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":68,"Direction":1}}
20:33:05 SHT: Shutter1 Real 29895, Start 27295, Stop 40000, Dir 1, Delay 0, Rtc 1.30 [s], Freq 1000
20:33:05 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":75,"Direction":1}}
20:33:06 SHT: Shutter1 Real 31895, Start 27295, Stop 40000, Dir 1, Delay 0, Rtc 2.30 [s], Freq 1000

As you can see the rule to switch off POWER1 is never send

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
ShutterOpen1
wait 2s
ShutterClose1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
POWER1 is disabled for atleast 1s when the direction changes

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

taste66, 2020-02-16:
I will try the solution indicated in this post: https://github.com/arendst/Tasmota/issues/7133#issuecomment-586680162
Will close the issue if it works!

taste66, 2020-02-16:
Can report 1000ms works. See console output:
`11:22:21 SHT: Shutter1 Real 11100, Start 0, Stop 40000, Dir 1, Delay 0, Rtc 5.55 [s], Freq 1000
11:22:21 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":28,"Direction":1}}
11:22:22 SHT: Shutter1 Real 13100, Start 0, Stop 40000, Dir 1, Delay 0, Rtc 6.55 [s], Freq 1000
11:22:22 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":33,"Direction":1}}
11:22:22 CMD: shutterclose1
11:22:22 SHT: Wait for Motorstop..
11:22:22 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-16T11:22:22","Uptime":"0T00:03:53","UptimeSec":233,"Vcc":3.433,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:22:22 MQT: stat/Shutter/01/RESULT = {"POWER1":"OFF"}
11:22:22 MQT: stat/Shutter/01/POWER1 = OFF (retained)
11:22:23 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-16T11:22:23","Uptime":"0T00:03:54","UptimeSec":234,"Vcc":3.432,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:22:23 MQT: stat/Shutter/01/RESULT = {"POWER1":"OFF"}
11:22:23 MQT: stat/Shutter/01/POWER1 = OFF (retained)
11:22:23 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-16T11:22:23","Uptime":"0T00:03:54","UptimeSec":234,"Vcc":3.430,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:22:23 MQT: stat/Shutter/01/RESULT = {"POWER2":"ON"}
11:22:23 MQT: stat/Shutter/01/POWER2 = ON (retained)
11:22:23 MQT: tele/Shutter/01/STATE = {"Time":"2020-02-16T11:22:23","Uptime":"0T00:03:54","UptimeSec":234,"Vcc":3.429,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"taste2","BSSId":"08:62:66:3D:BD:90","Channel":6,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:07"}}
11:22:23 MQT: stat/Shutter/01/RESULT = {"POWER1":"ON"}
11:22:23 MQT: stat/Shutter/01/POWER1 = ON (retained)
11:22:23 MQT: stat/Shutter/01/RESULT = {"ShutterClose1":0}
11:22:24 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
11:22:24 MQT: stat/Shutter/01/RESULT = {"SerialSend":"Done"}
11:22:24 SHT: Shutter1 Real 13385, Start 13700, Stop 0, Dir -1, Delay 0, Rtc 0.15 [s], Freq 1052
11:22:24 MQT: stat/Shutter/01/RESULT = {"Shutter1":{"Position":33,"Direction":-1}}
11:22:24 SHT: Shutter1 Real 12440, Start 13700, Stop 0, Dir -1, Delay 0, Rtc 0.60 [s], Freq 1052`

btw the reason I use an LC-TECH is it fits into a European style wall box if you cut the corners a bit.

taste66, 2020-02-16:
Looks like the rule is not trigger while     
delay(MOTOR_STOP_TIME);
is executed. As I am nog SW engineer I would not know how to fix this

stefanbode, 2020-02-16:
Yes, this might be correct, because it is a delay(500); that for sure freezes the program execution for 0.5 sec. There is no simple way to overcome this. You can set the delay in the INO file to "0". In this case, the motor immediately changes direction. If I give back control at this time I will have definitely side effects during the main execution of the functions.

taste66, 2020-02-17:
Hi Stefan, Thanks I will see if I can simply avoid you change the direction without a stop in between.
Currently this means te LC-TECH is not usable this way since it does break the direction relay (confirmed)

closed 2020-02-17 by taste66