stefanbode/Sonoff-Tasmota #221: upgrading from Sonoff-Tasmota 6.5.0.11 stb-1.5 by Theo Arends

adizag, 2021-11-10:
Hi @stefanbode 

I'm using your firmware for a very long time, but I'd like to upgrade to the main tasmota now after you migrated your great work in there.
Can you please advise abour upgrading method from Sonoff-Tasmota 6.5.0.11 stb-1.5 ?

tnx!

stefanbode, 2021-11-11:
Yes, please export your settings first. Then follow the upgrade procedure for like you have normal tasmota 6.5 version. There are some updates with the -minimal version required until you can go to version 10.

stefanbode, 2021-11-11:
Migrate to Sonoff-Tasmota 6.7.1 (http://ota.tasmota.com/tasmota/release-6.7.1/)
Migrate to Tasmota 7.2.0 (http://ota.tasmota.com/tasmota/release-7.2.0/)
--- Major change in parameter storage layout ---

Migrate to Tasmota 8.5.1 (http://ota.tasmota.com/tasmota/release-8.5.1/)
--- Major change in internal GPIO function representation ---

Migrate to Tasmota 9.1 (http://ota.tasmota.com/tasmota/release-9.1.0/)

adizag, 2021-11-14:
well... that didn't go well....
I upgraded to 6.7.1 then the switch became unavailable, not responsive to touch also... I believe I'll have to burn the firmware again....

adizag, 2021-11-14:
So I flashed from scratch, but im unable to restore from backup...
any idea?

[image]

adizag, 2021-11-15:
@stefanbode 
OK, I recovered the switch (phew, there were many changes from your fork to the main Tasmota!)
Now it is working nice.

One feature is missing and I couldn't find how to config it:
when the shutters are opening e.g. and I press the Down button it stops. I want the shutters to change direction in that case as it was in your fork.
How do I do that please?

adizag, 2021-11-17:
Solved it - I had to command `reset 6` before I started the upgrade.
Now everything is great.
tnx.

closed 2021-11-17 by adizag

stefanbode, 2021-11-17:
There a different command for every use case. You can assign buttons to the commands.  Take a look into the very long documentation