stefanbode/Sonoff-Tasmota #224: Shutter - inverse mode - slider not correct

gj52, 2023-11-16:
### PROBLEM DESCRIPTION
I use different shelly 2.5 for shutter control.
I had to set InverseMode in one installation.
All works fine, but the sider does not show correct open/close states.
Do I missing any option to set?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5 (not 2.5 PM - this works!)
- [X] Tasmota binary firmware version number used: 13.2.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Set SetOption80 1     # enable shutter and blinds support
Interlock 1,2 
Interlock ON
SwitchMode1 1
SwitchMode2 1
SetOption13 1
ShutterOpenDuration1  30
ShutterCloseDuration1 30
PowerOnState 0 # keep power(s) OFF after power up 
SetOption63  1 # Disable power state scanning at restart
SensorRetain 1
ShutterInvert1 1
WebButton1 ZU
WebButton2 AUF

### EXPECTED BEHAVIOUR
On the slider
open <-> close changes position
Shutter is open, but the slider shows close

### SCREENSHOTS
09:52:33.001 MQT: stat/wohn-rollo/STATUS = {"Status":{"Module":0,"DeviceName":"wohn-rollo","FriendlyName":["wohn-rollo-zu","wohn-rollo-auf"],"Topic":"wohn-rollo","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:52:33.008 MQT: stat/wohn-rollo/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.9:9541/data/firmwares/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"6T14:53:29","StartupUTC":"2023-11-09T17:59:04","Sleep":50,"CfgHolder":4617,"BootCount":144,"BCResetTime":"2022-09-15T12:13:20","SaveCount":1976,"SaveAddress":"F6000"}}
09:52:33.013 MQT: stat/wohn-rollo/STATUS2 = {"StatusFWR":{"Version":"13.2.0.2(tasmota)","BuildDateTime":"2023-11-09T14:35:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"416/699"}}
09:52:33.017 MQT: stat/wohn-rollo/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Saturn",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0020E209","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
09:52:33.031 MQT: stat/wohn-rollo/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
09:52:33.037 MQT: stat/wohn-rollo/STATUS5 = {"StatusNET":{"Hostname":"wohn-rollo","IPAddress":"192.168.0.244","Gateway":"192.168.0.1","Subnetmask":"255.255.254.0","DNSServer1":"192.168.0.4","DNSServer2":"0.0.0.0","Mac":"4C:75:25:34:5B:3C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:52:33.041 MQT: stat/wohn-rollo/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.9","MqttPort":1883,"MqttClientMask":"wohn-rollo","MqttClient":"wohn-rollo","MqttUser":"muecke","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:52:33.049 MQT: stat/wohn-rollo/STATUS7 = {"StatusTIM":{"UTC":"2023-11-16T08:52:33","Local":"2023-11-16T09:52:33","StartDST":"2023-03-26T02:00:00","EndDST":"2023-10-29T03:00:00","Timezone":"+01:00","Sunrise":"08:00","Sunset":"17:08"}}
09:52:33.057 MQT: stat/wohn-rollo/STATUS10 = {"StatusSNS":{"Time":"2023-11-16T09:52:33","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.0},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"TempUnit":"C"}}
09:52:33.064 MQT: stat/wohn-rollo/STATUS11 = {"StatusSTS":{"Time":"2023-11-16T09:52:33","Uptime":"6T14:53:29","UptimeSec":572009,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Saturn","BSSId":"2C:91:AB:83:B8:DC","Channel":1,"Mode":"11n","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}
09:52:33.068 MQT: stat/wohn-rollo/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":300,"Close":300,"50perc":50,"Delay":0,"Opt":"0001","Calib":[300,500,700,900,1000],"Mode":"1"}}}

[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2023-11-16:
Can we check some generell fist. 
Shutterinvert is 1
Shutteropen command and shutterclose command work as expected 
Shutterclose report 100
Shutteropen Report 0

Yes on invert the open and close text near the slider switches position to reflect that now 0 is open

What do you expect.

stefanbode, 2023-11-16:
You write this works on 2.5 PM and not on 2.5. I assume one is esp32 and the other esp8266 and the error might be in the old 8266 version.

gj52, 2023-11-16:
OK, I will try with a new Shelly 2.5PM

closed 2023-11-16 by gj52

stefanbode, 2023-11-17:
I checked the code and cannot find a reason why open and close do not swap positions. Currently cannot test it. Will do a check next week. Can you answer in the questions? Will make my debugging easier. Thx

stefanbode, 2023-11-20:
@gj52 : I checked with ESP8266 and also here the invert does work as expected. "Close" is on the left when not inverted and on the right when inverted. This is in line with specification